ARM-software/vscode-cmsis-csolution #1: Cant convert a Keil MDK project to a csolution project

liusj5257, 2022-09-21:
After  using the file picker that displays, select the .uvprojx that I want to convert and click Select.
Then check the OUTPUT tab, it only shows “GrpcServiceError: 14 UNAVAILABLE 14 UNAVAILABLE: No connection established”

micque01, 2022-09-21:
Hi @liusj5257, there's a known issue with one of the underlying tools which will be fixed in an upcoming release. In the meantime could you please share which operating system you are using? I could provide a potential workaround for the problem. Thanks!

liusj5257, 2022-09-21:
> Hi @liusj5257, there's a known issue with one of the underlying tools which will be fixed in an upcoming release. In the meantime could you please share which operating system you are using? I could provide a potential workaround for the problem. Thanks!

Windows 11 pro 21H2
22000.856
1000.22000.856.0

micque01, 2022-09-21:
There should be a directory in your Windows home directory called `AppData/Local/cmsis-core-tools`. Please delete that directory and try reloading VS Code then trying the conversion again. Sorry about this issue!

liusj5257, 2022-09-21:
> 
I tried but it dont work for me，and still shows 
"GrpcServiceError: 14 UNAVAILABLE 14 UNAVAILABLE: No connection established"

micque01, 2022-10-04:
Hi @liusj5257, could you try updating to the latest version of the csolution extension and reloading vscode?

liusj5257, 2022-10-08:
> 

I try the latest version but still the same issue.

escherstair, 2022-11-29:
@micque01 I can't convert an existing MDK project to csolution, but with a different error.
If I run **CMSIS: Convert MDK project to csolution** command from an empty folder, I get the error
`Error: Can't map 'd:\deleteme\d\Projects\Blinky\Blinky.uvprojx': it has no overlap with '\.' or '\.'`
If I run it from a csolution (as an example the Demo_EW suggested [here](https://github.com/ARM-software/vscode-cmsis-csolution#work-with-a-csolution-example-project)), the erro is
`Error: Can't map 'd:\deleteme\d\Projects\Blinky\Blinky.uvprojx': it has no overlap with 'd:\deleteme\cmsis_solution\Demo_EW' or 'd:\deleteme\cmsis_solution\Demo_EW'`

Windows 10

mcgordonite, 2022-11-30:
Hi @escherstair, thanks for reporting that. I think it is a separate issue, so I created https://github.com/ARM-software/vscode-cmsis-csolution/issues/4. It will be fixed in the next release.

dbzHTK, 2023-01-30:
Hi I do have also e different Issue in converting MDK project to a csolution one. I got :
```
<Target> "DEBUG" <umfTarg> scatter file generation not supported
<Target> "Release" <umfTarg> scatter file generation not supported
```
Im Using: 

Windows 10 Pro 10.0.19044
MDK 5.34.0.0

Thank you in advance!

mcgordonite, 2023-02-01:
Hi @dbzHTK, please see the comment here: https://github.com/ARM-software/vscode-cmsis-csolution/issues/6#issuecomment-1339755050.

ccli0, 2023-03-31:
After using the file picker that displays, select the .uvprojx that I want to convert and click Select.
Then check the OUTPUT tab, it only shows “GrpcServiceError: 14 UNAVAILABLE 14 UNAVAILABLE: No connection established”
-> How to fix this problem?

mcgordonite, 2023-03-31:
> Then check the OUTPUT tab, it only shows “GrpcServiceError: 14 UNAVAILABLE 14 UNAVAILABLE: No connection established”

Hi @ccli0, we've seen this issue for users working behind a proxy. Is that the case for you? If so, see the "Configure an HTTP Proxy" section in https://github.com/ARM-software/vscode-cmsis-csolution#install-the-cmsis-toolbox.

ccli0, 2023-03-31:
Could you give me a practical example of how to set up the PROXY environment variable?
It's not very clear after reading the description.

Are there other possible causes of this issue if user does not work behind a proxy?

mcgordonite, 2023-03-31:
How to set an environment variable depends on what operating system you are using. For example on Windows you can use the control panel to do so. I'd suggest looking for a guide for your operating system. The exact value to set will depend on the proxy set up you have locally, but `HTTP_PROXY` and `HTTPS_PROXY` should be set to a URL like `https://example.com:8000`. `NO_PROXY` should be set to `localhost,127.0.0.1` to exclude localhost.

> Are there other possible causes of this issue if user does not work behind a proxy?

This error appears when the extension has a problem communicating with the "cmsis" process that it uses to retrieve CMSIS packs content. This could happen for many reasons. It might help to know your operating system, and you could share the logs in the developer tools (Command Palette > Toggle Developer Tools in VS Code). Look for any lines relating to cmsis csolution, or issues spawning sub processes.

ccli0, 2023-04-07:
Hi @mcgordonite,

My operating system is Windows.
I got the same errors with the following two commands.
CMSIS: Convert MDK project to csolution
CMSIS: Install required packs for active solution

In "CMSIS: Convert MDK project to csolution" case, there are no message in the console (Toggle Developer Tools in VS Code)
In "CMSIS: Install required packs for active solution" case, some output message shown bellow

14 UNAVAILABLE 14 UNAVAILABLE: No connection established
c @ workbench.desktop.main.js:750
(anonymous) @ workbench.desktop.main.js:750
invoke @ workbench.desktop.main.js:86
deliver @ workbench.desktop.main.js:86
fire @ workbench.desktop.main.js:86
addNotification @ workbench.desktop.main.js:750
notify @ workbench.desktop.main.js:1691
(anonymous) @ workbench.desktop.main.js:1586
d @ workbench.desktop.main.js:1586
$showMessage @ workbench.desktop.main.js:1586
N @ workbench.desktop.main.js:1593
M @ workbench.desktop.main.js:1593
H @ workbench.desktop.main.js:1593
G @ workbench.desktop.main.js:1593
(anonymous) @ workbench.desktop.main.js:1593
invoke @ workbench.desktop.main.js:86
deliver @ workbench.desktop.main.js:86
fire @ workbench.desktop.main.js:86
fire @ workbench.desktop.main.js:556
G.onmessage @ workbench.desktop.main.js:1691

Is there anything else I need to check further?

mcgordonite, 2023-04-11:
Glad to hear it @ccli0. 

In that case I will close this issue. There are other issues with importing Keil MDK projects that are currently being worked on (e.g.: https://github.com/ARM-software/vscode-cmsis-csolution/issues/6#issuecomment-1339755050), but I believe the "14 UNAVAILABLE 14 UNAVAILABLE: No connection established" error is resolved.