ARM-software/vscode-cmsis-csolution #2: Can't setup in ubuntu, instructions unclear

fredo514, 2022-10-06:
I tried to install in Ubuntu with GCC and couldn't get it to work.

Everything installed without errors, including getting the CMSIS packs using cpackget, and the CMSIS extension showed the project settings correctly. But the Project Manager output keep saying that the packs are missing, and the .cprj is not created. Building fails because the .cprj cannot be found.

Note: It would be nice to add a bit more details on how to set the environment variables, or at least to make the change permanent.
Note: It would be nice to mention that the command to get packs should be run in the terminal and not the command palette.

mcgordonite, 2022-11-24:
@escherstair apologies for the slow response on this.

The extension gets the location of the pack cache from the `CMSIS_PACK_ROOT` environment variable. This can be overridden by the `packCachePath` setting in VS Code. Otherwise, it falls back to a default path. Some things to check:
- Is the environment variable and/or setting set up correctly to point at the packs folder on your system?
- Can you run `csolution convert -s path/to/solution` successfully from the built-in terminal in VS Code?

> Note: It would be nice to add a bit more details on how to set the environment variables, or at least to make the change permanent.

This can be different for each linux distro, so not really practical to document here.

> Note: It would be nice to mention that the command to get packs should be run in the terminal and not the command palette.

Great suggestion, we'll update that for the next release.

mcgordonite, 2022-11-28:
>> Note: It would be nice to mention that the command to get packs should be run in the terminal and not the command palette.
>
> Great suggestion, we'll update that for the next release.

That's updated now.

mcgordonite, 2022-12-07:
Closing this issue. @fredo514 please open a separate issue if you are still having problems with configuring the CMSIS_PACK_ROOT.

closed 2022-12-07 by mcgordonite