ARM-software/vscode-cmsis-csolution #5: YAML solution file error

lmdanny, 2022-12-01:
Type: Bug Report

**Describe the bug**
- OS and Version: WIN11 x64
- VS Code Version: latest
- Extension Version: latest
- Target Device: n.a.
- Other extensions you installed (and if the issue persists after disabling them): NO, clean vscode with CMSIS extentions and other required things according to documentation (I guess)
- A clear and concise description of what the bug is.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Suppose no errors? maybe I missed something. 

**Code sample and logs**
- Code sample
- `tasks.json`

**Screenshots**

[image]

**Additional context**
Have configured everything to work with tools and compiler shipped with MDK5.37, all paths environment variables were configured according to installation manual

mcgordonite, 2022-12-01:
Hi @Imdanny, the issue here is the extension is using the `--load` option when converting csolutions to CPRJ, which was added in csolution version 1.3.0. I think that's later than the version shipped in MDK 5.37.

One fix would be to install the latest release of CMSIS Toolbox from here: https://github.com/Open-CMSIS-Pack/cmsis-toolbox/releases.

Alternatively, you could disable the automatic conversion from the preferences and just convert manually by running `csolution convert` from the command line.

lmdanny, 2022-12-02:
Hi, that did work indeed

I was a bit confused because the instructions somewhere said that version 5.36 was sufficient.

Thank anyway, I’ll just keep playing with it for a while.

Danny

closed 2022-12-02 by mcgordonite