ARM-software/vscode-cmsis-csolution #6: Import MDK Project features

lmdanny, 2022-12-02:
Type: Feature Request

Import uvprojx needs support for scatter files and other common things. I have a lot of projects that were created with MDK together with Cube MX. 

So:

1) This is what I got on trying to convert a very basic project
<Target> "Target 1" <umfTarg> scatter file generation not supported
<gpdscs> are not supported

2) I do not know the status for CubeMX integration, (I do see it in the component list), but I do not see how I can do anything with it. 
maybe I'm just overlooking things but it is really a must have.

fredo514, 2022-12-06:
It's a bit annoying, but you can fix it manually by unticking "Use Memory Layout from Target Dialog" in the Target Options dialog under the Linker tab in uVision. Then you build and that will generate a scatter file.

mcgordonite, 2022-12-08:
> I do not know the status for CubeMX integration, (I do see it in the component list), but I do not see how I can do anything with it. maybe I'm just overlooking things but it is really a must have.

This definitely is a must have, and we've got it on our roadmap for both this VS Code extension and in Open-CMSIS-Pack. It will be supported with the CMSIS generators feature if you come across that in any documentation.

escherstair, 2022-12-12:
Another CubeMX unsupported feature is GPDSCS configuration file.
In the past someone started developing a conversion [python script](https://groups.google.com/g/alt.sources/c/uxOBz90oNqk/m/rdhWIRW1GwAJ).
@mcgordonite do you need help in some way to speed up the development of this feature?

JonTBeckett, 2023-03-28:
I'm looking into using GPDSC files in a Keil project I'm working on and hope to migrate to csolution with the project in the future. I'd love to see GPDSC files supported for sure!

Just wanted to voice my support for these features. 

Thanks!

mcgordonite, 2023-06-12:
The latest release should add support for converting uvprojx files that reference scatter files.