ARM-software/vscode-cmsis-csolution #7: Invisible buttons till hover are a genuine annoyance

lmdanny, 2022-12-02:
Type: Feature Request

Wouldn't it be a good idea to either have the buttons for creating projects or changing components to be always visible, and if this cannot be done just move all the controls down and just add in a simple toolbar above all the other items.

Now I do know where to find them, but it wasn't easy :)

[image]

lmdanny, 2022-12-06:
Okay, just reading through there seems this is already open since 2020

https://github.com/microsoft/vscode/issues/109030

That gives me little hope that this will be fixed or changed soon.

For users, I believe these particular features such as the component selection and perhaps also other functionality that is built in MDK currently and might come to the vscode extensions (I don't know the roadmap), a separate section with would not be a harm to the user's experience.

mcgordonite, 2022-12-07:
Agreed, unless our team can contribute a fix up to VS Code I don't expect to see it fixed soon.

We'll investigate different options to improve the UX here.

kanthanp, 2023-07-19:
This usability has now been fixed.

mcgordonite, 2023-09-04:
Closing: the fix in https://github.com/microsoft/vscode/issues/109030 is now live in VS Code.

closed 2023-09-04 by mcgordonite