ARM-software/vscode-cmsis-csolution #8: I've a problem of required pack in the csolution

Osika91, 2022-12-03:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11 
- VS Code Version: 1.73.1
- Extension Version: Arm CMSIS csolution v0.8.1
- Target Device: STM32F103RB

- I'm trying to build and flash a program on a Nucleo STM32F103RB. I'm using vscode and Arm CMSIS csolution. The problem is that the pack are installed on the location %HOME%\AppData\Local\Arm\Packs but my csolution project doesn't find them even if the folder of the pack required are in my project. Have you a solution for this problem ? I think I have forgot to do something but I don't find what.

**Screenshots**

[image]
You can see this screenshot that I've problen on required pack in the terminal when i want to run the command : 

`csolution convert -s Blinky.csolution.yml`

mcgordonite, 2022-12-05:
The extension gets the location of the pack cache from the CMSIS_PACK_ROOT environment variable. This can be overridden by the packCachePath setting in VS Code. Otherwise, it falls back to a default path.

- Is the CMSIS_PACK_ROOT set correctly? You could check by echoing the value using VS Code's built in terminal.
- Are you overriding the environment variable with the packCachePath setting in VS Code's settings?

mcgordonite, 2022-12-05:
Have you tried to build the project? Do you see the same error as highlighted in VS Code? If not, this could be a problem with intellisense.

Otherwise, this sounds like a question for the Open-CMSIS-Pack community here: https://github.com/Open-CMSIS-Pack/devtools/issues. This repo is specifically for the VS Code extension rather than general csolution support.

Osika91, 2022-12-05:
Yes I build the project and there is different error. I'm going to ask my question the Open-CMSIS-Pack community. Thanks you for your help !

closed 2022-12-05 by Osika91