ARM-software/vscode-cmsis-csolution #10: Daemon Isn't Starting

vindicatorr, 2022-12-06:
Sandboxed: No
**Describe the bug**
- OS and Version: Arch Linux
- VS Code Version: 1.72.0 (OSS)
- Extension Version: v0.8.1
- Target Device: STM32F4-Discovery
- Other extensions you installed (and if the issue persists after disabling them): N/A
- A clear and concise description of what the bug is. GRPC error. I'm guessing daemon isn't running, and is needed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the `cmsis` extension
2. Click on '+' (create a solution)
3. See error

**Expected behavior**

The daemon should run. It appears not to be.

**Code sample and logs**
- Code sample N/A
- `tasks.json` N/A
- daemon start log (plain):
```
time="2022-12-06T11:32:42-06:00" level=error msg="server failed to listen: listen tcp: lookup localhost on 1.1.1.1:53: no such host"
```

- daemon start log (-s 127.0.0.1)
```
time="2022-12-06T11:35:44-06:00" level=info msg="Server is listening with configuration" BackupDownloads=false InitCacheDir=true IsPublic=true address="127.0.0.1:9000"
```

- daemon log after extension comms
```
time="2022-12-06T11:36:43-06:00" level=debug msg="Loaded pdsc file." name=nRF_SoftDevice_S130 vendor=NordicSemiconductor version=2.0.0-7.alpha
time="2022-12-06T11:36:43-06:00" level=debug msg="Loaded pdsc file." name=STBlueNRG-1_DFP vendor=Keil version=1.2.0
...
time="2022-12-06T11:36:51-06:00" level=warning msg="Board definition found in 17 packs" board="NordicSemiconductor, nRF52 PCA10036, V1.0.0" packList="[{NordicSemiconductor nRF_Serialization 2.1.0} {NordicSemiconductor nRF_SoftDevice_S110 8.0.3} {NordicSemiconductor nRF_SoftDevice_S120 2.1.1} {NordicSemiconductor nRF_SoftDevice_S210 5.0.2} {NordicSemiconductor nRF_SoftDevice_S310 3.0.1} {NordicSemiconductor nRF_ANT 2.0.1-2.alpha} {NordicSemiconductor nRF_BLE 4.0.0-2.alpha} {NordicSemiconductor nRF_Drivers 4.0.0-2.alpha} {NordicSemiconductor nRF_Drivers_External 1.2.1-2.alpha} {NordicSemiconductor nRF_Examples 11.0.0-2.alpha} {NordicSemiconductor nRF_Libraries 4.0.0-2.alpha} {NordicSemiconductor nRF_NFC 1.0.0-2.alpha} {NordicSemiconductor nRF_Properitary_RF 1.2.3-2.alpha} {NordicSemiconductor nRF_RTX 4.74.0-9} {NordicSemiconductor nRF_SoftDevice_Common 2.0.0-2.alpha} {NordicSemiconductor nRF_SoftDevice_S130 2.0.0-7.alpha} {NordicSemiconductor nRF_SoftDevice_S132 2.0.0-7.alpha}]" sourcePack="NordicSemiconductor, nRF_Serialization, 2.1.0"
time="2022-12-06T11:36:51-06:00" level=warning msg="Board definition found in 17 packs" board="NordicSemiconductor, N5 Starter Kit, " packList="[{NordicSemiconductor nRF_Serialization 2.1.0} {NordicSemiconductor nRF_SoftDevice_S110 8.0.3} {NordicSemiconductor nRF_SoftDevice_S120 2.1.1} {NordicSemiconductor nRF_SoftDevice_S210 5.0.2} {NordicSemiconductor nRF_SoftDevice_S310 3.0.1} {NordicSemiconductor nRF_ANT 2.0.1-2.alpha} {NordicSemiconductor nRF_BLE 4.0.0-2.alpha} {NordicSemiconductor nRF_Drivers 4.0.0-2.alpha} {NordicSemiconductor nRF_Drivers_External 1.2.1-2.alpha} {NordicSemiconductor nRF_Examples 11.0.0-2.alpha} {NordicSemiconductor nRF_Libraries 4.0.0-2.alpha} {NordicSemiconductor nRF_NFC 1.0.0-2.alpha} {NordicSemiconductor nRF_Properitary_RF 1.2.3-2.alpha} {NordicSemiconductor nRF_RTX 4.74.0-9} {NordicSemiconductor nRF_SoftDevice_Common 2.0.0-2.alpha} {NordicSemiconductor nRF_SoftDevice_S130 2.0.0-7.alpha} {NordicSemiconductor nRF_SoftDevice_S132 2.0.0-7.alpha}]" sourcePack="NordicSemiconductor, nRF_Serialization, 2.1.0"
time="2022-12-06T11:36:51-06:00" level=debug msg="Memory usage after processing boards" Alloc (MiB)=76 NumGC=27 Sys (MiB)=190 TotalAlloc (MiB)=596
time="2022-12-06T11:36:51-06:00" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=ListBoards grpc.service=packs.Packs grpc.start_time="2022-12-06T11:36:51-06:00" grpc.time_ms=88.72 span.kind=server system=grpc
```

**Screenshots**

**Additional context**

Error:
```
Solution service failure: 14 UNAVAILABLE 14 UNAVAILABLE: No connection established GrpcServiceError: 14 UNAVAILABLE 14 UNAVAILABLE: No connection established at Object.create (/home/userName/.vscode-oss/extensions/arm.cmsis-csolution-0.8.1-linux-x64/dist/desktop/extension.js:2:1164258) at Object.parse (/home/userName/.vscode-oss/extensions/arm.cmsis-csolution-0.8.1-linux-x64/dist/desktop/extension.js:2:1164405) at a.<anonymous> (/home/userName/.vscode-oss/extensions/arm.cmsis-csolution-0.8.1-linux-x64/dist/desktop/extension.js:2:1163511) at a.emit (node:events:526:28) at Object.onReceiveStatus (/home/userName/.vscode-oss/extensions/arm.cmsis-csolution-0.8.1-linux-x64/dist/desktop/extension.js:2:336379) at Object.onReceiveStatus (/home/userName/.vscode-oss/extensions/arm.cmsis-csolution-0.8.1-linux-x64/dist/desktop/extension.js:2:329495) at /home/userName/.vscode-oss/extensions/arm.cmsis-csolution-0.8.1-linux-x64/dist/desktop/extension.js:2:289349 at process.processTicksAndRejections (node:intern...
```

I looked at the extension directory and saw the `cmsis` tool and explored that.
I saw there was a "daemon" feature, which made me think it should be running. It was not. No `cmsis` process, nor any port 9000 being used.
So I tried to `start` it manually, but still saw no port.
Then I added verbosity and saw it say it started and how to stop it.
Then I added the logging and saw mention of address `1.1.1.1`. Odd. Why?
I also saw `--help` mention the .config, but the path mentioned as `default` does not exist.
So when I manually started the process with the address:port 127.0.0.1:9000, it ran, and the "create a solution" began with populating the device list.

mcgordonite, 2022-12-07:
Hi @vindicatorr, thanks for your detailed bug report.

The daemon tries to start on hostname "localhost" by default. Is it possible your system doesn't have "localhost" configured. Is there an entry for it in /etc/hosts?

vindicatorr, 2022-12-07:
Correct, my `hosts` file is blank.

madchutney, 2022-12-07:
@vindicatorr Thanks for the bug report and for the investigation. The `cmsis` tool is actually a gRPC server that should indeed be running and as you worked out it should normally be run on `localhost`.

I think that could be the cause of the problem, could you add `127.0.0.1 localhost` and retry?

mcgordonite, 2022-12-07:
With `cmsis-csolution.coreTools.address` the extension won't start the daemon itself, it will assume there is one already started. If `cmsis-csolution.coreTools.path` is set, it looks at that path for the daemon binary.

What happens if you remove both settings and try again? You may need to restart VS Code to see the change unfortunately.

Thanks for your help investigating this.

closed 2022-12-07 by vindicatorr

mcgordonite, 2022-12-07:
> Another thought, if cmsis-csolution.coreTools.path is expecting the daemon to be found there, then maybe alert the user the path is incorrect when the cmsis binary is not found there (or more clarity of what coreTools is (aka, the extension bin)

Good suggestion, I've added it to our backlog. Thanks again for investigating this.

madchutney, 2022-12-07:
@vindicatorr Many thanks, your time and effort is much appreciated.