ARM-software/vscode-cmsis-csolution #11: Don't see the + or options to create new project

fredo514, 2022-12-06:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows_NT x64 10.0.19044
- VS Code Version: 1.73.1
- Extension Version: v0.3.6
- Target Device: STM32F415RGTx
- Other extensions you installed (and if the issue persists after disabling them): Keil Studio Pack
- In the CMSIS tab, I don't see the the Create a solution nor the Manage software components icons

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the CMSIS tab
2. See error

**Expected behavior**
In the CMSIS tab, see the the Create a solution nor the Manage software components icons, per the readme.

mcgordonite, 2022-12-07:
Hi @fredo514, both the create solution and manage components views are marked as "experimental features". Please enable the setting as described here: https://github.com/ARM-software/vscode-cmsis-csolution#create-a-csolution-project.

Note: the manage components view will no longer be marked experimental from the next release.

fredo514, 2022-12-07:
Oops, missed it. Sorry about that.