ARM-software/vscode-cmsis-csolution #15: C/C++ language standard settings are not properly converted from Keil MDK project

Overlap0190, 2022-12-22:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11 22H2 (Build 22621.963)
- VS Code Version: v1.74.2
- Extension Version: v0.9.0
- Target Device: Nuvoton M251
- Description: When using "Convert MDK project to csolution" the C/C++ language standard compiler flags are not set properly.

My settings in Keil MDK:
[image]

After the conversion the *.csolution.yaml looks like this:
````
misc:
  - C*:
      - -fno-rtti
      - -funsigned-char
      - -fshort-enums
      - -fshort-wchar
      - -ffunction-sections
      - -std=c99
      ...
````
The C standard is applied to C/C++ files. The C++ standard is completely missing.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this repository https://github.com/RobertRostohar/Demo_EW/
2. Open the EdgeImpulse folder in VS Code
3. Convert the example.uvprojx to csolution
4. Open the generated example.csolution.yaml and the misc segment should look like mentioned above.

**Expected behavior**
I would expect something like this in the *.csolution.yaml:
````
misc:
  - C:
      - -std=c99
    CPP:
      - -std=c++14
    C*:
      - -fno-rtti
      - -funsigned-char
      - -fshort-enums
      - -fshort-wchar
      - -ffunction-sections
      ...
````

JoeTheGuitarist, 2023-01-03:
@ducnguyenarm FYI

mcgordonite, 2023-09-04:
Closing this: C++ support was added to the µVision converter in version 1.6.0.

closed 2023-09-04 by mcgordonite