ARM-software/vscode-cmsis-csolution #17: Failed to load solution at c:\Practice\Demo_EW\Blinky\Blinky.csolution.yml

UniverGao, 2023-01-11:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10 Pro 21H2
- VS Code Version: 1.74.3(user setup)
- Extension Version: v0.10.0
- Target Device: Example project in 'https://github.com/RobertRostohar/Demo_EW/'
- An error of Arm CMSIS solution extention: 
- `Failed to load solution at 
   c:\Practice\Demo_EW\Blinky\Blinky.csolution.yml`  
   when the project folder was opened in vs code, and 'Manage Software Components' interface of the CMSIS solution extention was empty. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install the cmsis tools
2. Install ninja and cmake
3. Install the pack that the project need
4. Open the folder of the project

**Expected behavior**

No error pop-out and 'Manage Software Components' interface of the CMSIS solution extention shows the pack components existed.

**Code sample and logs**
- Output of CMSIS Project Manager
`csolution convert --load required -s c:\Practice\Demo_EW\Blinky\Blinky.csolution.yml --no-check-schema
C:/Practice/Demo_EW/Blinky/Blinky.Debug+AVH_MPS3_Corstone-300.cprj - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Debug+B-U585I-IOT02A.cprj - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Release+AVH_MPS3_Corstone-300.cprj - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Release+B-U585I-IOT02A.cprj - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.cbuild-idx.yml - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Debug+AVH_MPS3_Corstone-300.cbuild.yml - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Debug+B-U585I-IOT02A.cbuild.yml - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Release+AVH_MPS3_Corstone-300.cbuild.yml - info csolution: file generated successfully
C:/Practice/Demo_EW/Blinky/Blinky.Release+B-U585I-IOT02A.cbuild.yml - info csolution: file generated successfully`

**Screenshots**

[image]
[image]

**Additional context**

mcgordonite, 2023-01-11:
Hi @UniverGao, to help debug this it would be useful to have the logs from VS Code:
1. Open the command palette (CTRL-SHIFT-P on Windows).
2. Run the "Developer: Toggle Developer Tools" command. The dev tools panel should appear.
3. Open the Console tab.
4. Copy the logs, or right click anywhere in the Console tab and select "Save As".

UniverGao, 2023-01-12:
I did the test and the log is here:
Best regrards.

[vscode-app-csolution-v0.10.0.log](https://github.com/ARM-software/vscode-cmsis-csolution/files/10398527/vscode-app-csolution-v0.10.0.log)

UniverGao, 2023-01-13:
@mcgordonite 
Sorry that the extention was not installed correctly. I reinstalled the extension, cmsis.exe could be run corrently, the error message 'Failed to load solution at...' was disappeared, but the 'Manage Software Components' interface of the CMSIS solution extention was still empty.
PS: 
- The CPU type of my computer is i5-1145G7(Intel), x86-64
- The extension was also installed in WSL2(Windows subsystem for Linux), and exists she same problem. 

The log shown below is about the status after the extension correctly re-installed:
[vscode-app-cmsis-win-v0.10.0.log](https://github.com/ARM-software/vscode-cmsis-csolution/files/10409061/vscode-app-cmsis-win-v0.10.0.log)

mcgordonite, 2023-01-16:
It looks like the issue is that csolution example includes components via layers. The preview Manage Components view doesn't support layers yet, although basic support is in the roadmap for the next few months. If you'd like to try out the component manager, you could use an example that doesn't use layers, e.g. https://github.com/Open-CMSIS-Pack/csolution-examples/.

closed 2023-01-18 by UniverGao