ARM-software/vscode-cmsis-csolution #18: Add package that is not

UniverGao, 2023-01-24:
Type: Feature Request

UniverGao, 2023-01-24:
There is so many MCU products that keil administrator cannot consider every kind of SoC product. So some mauufactor of chip offer *.pack file whitch is not exist in the keil list. So is there any way to add local pack file?

mcgordonite, 2023-01-31:
Hi, the csolution file supports referencing a local file. See `path` in the docs here: https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Input-Format.md#pack-selection. For example:
```yaml
packs:
  - pack: NXP::K32L3A60_DFP
    path: ./local/NXP/K32L3A60_DFP      # with path to the pack (local copy, repo, etc.)
```
The CMSIS toolbox should fully support this, so you should be able to build csolution files referencing local packs.

However, this VS Code extension doesn't support this yet, so CMSIS software components referenced in local packs won't appear in the Software Components UI yet. That is on the roadmap.

mcgordonite, 2023-06-12:
Hi @UniverGao, local pack files should now be supported in this extension.

Run `cpackget add /path/to/file.pdsc` from the command line to register the pdsc file with your pack cache, reload VS Code, then you should see components from the pack appearing in the component manager. We're working on removing the need for reloading VS Code.

Hope that helps!

closed 2023-06-12 by mcgordonite