ARM-software/vscode-cmsis-csolution #19: .cmake changed structure in new version

ManInNight, 2023-01-25:
Hello,
I am not sure from the instructions if the information given for Windows is correct, because the template .cmake file has changed from what is given in the instructions:

############### EDIT BELOW ###############
set(TOOLCHAIN_ROOT "C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\12.2 rel1\bin")
set(TOOLCHAIN_VERSION "12.2.1")
set(PREFIX arm-none-eabi-)
set(EXT .exe)

Are these settings correct for Windows?

mcgordonite, 2023-01-31:
Hi, thanks for the feedback. That looks correct, if you have installed arm-none-eabi-gcc in that directory.

We will update the README to match the new template in the next release.

mcgordonite, 2023-02-08:
This was updated in release 0.12.0.

closed 2023-02-08 by mcgordonite