ARM-software/vscode-cmsis-csolution #20: cmsis-csolution.getBinaryFile command not returning the correct path when csolution.yml outdir is not default

LMESTM, 2023-02-06:
The ticket short description as itself seems related to #12 but it is maybe another corner case.

The csolution.yml project defines an "outdir" option:  
https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Input-Format.md#output-dirs

When using default output dir:
` outdir: ./out/$Project$/$TargetType$`
there seems to be no issue and vscode-cmsis-csolution can correctly find the binary **when trying to flash**.

But when outdir is specified in the csolution.yml project file like below:
`outdir: $BuildType$`
then 
The build part is correctly updated, but vscode-cmsis-csolution **can't find the binary when trying to flash** and displays error like:
ENOENT: no such file or directory, open '[..]\out\test\NUCLEO-U575ZI-Q\GCC\test.GCC+B-U585I-IOT02A.axf'

Possibly this is related to "command:cmsis-csolution.getBinaryFile" ?

**To Reproduce**
Steps to reproduce the behavior:
1. Get your default csolution.yml project file and add options like:
```
  output-dirs:
    intdir: $BuildType$
    outdir: $BuildType$
```

**Expected behavior**
The vscode-cmsis-csolution extension, or possibly "command:cmsis-csolution.getBinaryFile", should take into account this option when looking for the binary file to flash.

System info
- OS and Version: Windows 10
- VS Code Version: 1.75.0
- Extension Version: 0.11.1
- Target Device: NUCLEO_U575ZI
- Other extensions you installed (and if the issue persists after disabling them): installed as part of Keil Studio Pack

mcgordonite, 2023-02-07:
> Possibly this is related to "command:cmsis-csolution.getBinaryFile" ?

That's correct, it looks like the `getBinaryFile` command from this extension doesn't take `output-dirs` into account. We'll fix this in an upcoming release.

For now, you could hardcode the path to the binary file in your task configuration rather than using getBinaryFile.

LMESTM, 2023-02-20:
@mcgordonite  one extra comment, it also looks like `getBinaryFile`  doesn't manage that the binary file would be a .ef file in case of GCC build type, and not an .axf

mcgordonite, 2023-02-20:
Thanks @LMESTM, added to our backlog. The extension will be updated to get this information from the underlying CMSIS toolbox tools, so all of these cases should be handled in future.

LMESTM, 2023-02-20:
@mcgordonite  sounds good, thanks for confirmation  ! 
Out of curiosity: any idea of a timeframe ?

mcgordonite, 2023-03-01:
This was less easy to solve than anticipated, we need additional information from csolution to compute this. I've raised https://github.com/Open-CMSIS-Pack/devtools/issues/778 to discuss. Once that is resolved I expect it will be trivial to implement the fix in the extension.

jkrech, 2023-03-01:
Workaround: When using cmsis-toolbox 1.5.0 I suggest to add to the cproject.yml

  output:
    - type: elf
      file: $BuildType$+$TargetType$.axf
      
This should override and create the axf file extension even for GCC and I would assume the debugger would find and load the file.

mcgordonite, 2023-03-01:
Adding to @jkrech's workaround, you also need to include the `$PROJECT` in the file for now:
```yaml
output:
    - type: elf
      file: $Project$.$BuildType$+$TargetType$.axf
```
The next release of this extension will update to the new default binary path without $PROJECT though, so it might be worth waiting for that.

LMESTM, 2023-03-03:
Thanks for your support, we'll try that out !

jeromecoutant, 2023-03-03:
> Adding to @jkrech's workaround, you also need to include the `$PROJECT` in the file for now

I think first comment was OK,
with 1.4.0 default name $Project$.$BuildType$+$TargetType$
with 1.5.0 default name $BuildType$+$TargetType$

mcgordonite, 2023-03-08:
Hi @LMESTM, version 0.16.1 of the extension released today uses an improved `getBinaryFile` command so you won't need the `$PROJECT.` at the start of the output config:
```yaml
output:
  - type: elf
    file: $BuildType$+$TargetType$.axf
```

I'm leaving this issue open until the extension uses the cbuild.yml file to back `getBinaryFile`, so we get full support for csolution's output configuration.

LMESTM, 2023-03-08:
ok thanks @mcgordonite

micque01, 2023-03-20:
Hi @LMESTM, in the latest version of the vscode-cmsis-csolution extension, the `getBinaryFile` command now takes into account the `output-dirs` configuration from the *.csolution.yml file.

Let us know if you have any further issues!

VVESTM, 2023-04-04:
Hi, tested with success on my side. I was able to use getBinaryFile with an output-dirs option and without any patch to generate .axf file. So, sounds good !! 
Thanks for your support.

LMESTM, 2023-04-04:
thanks @VVESTM for testing as I couldn't find time for it 

@mcgordonite @micque01  do you want us to close the ticket ?

mcgordonite, 2023-04-04:
Thanks for testing this out!

closed 2023-04-04 by mcgordonite