ARM-software/vscode-cmsis-csolution #21: Does Build Clean really clean ?

LMESTM, 2023-03-07:
Not sure if this is a bug or a request for enhancement

**Describe the bug**
- OS and Version: Windows
- VS Code Version:  1.76.0
- Extension Version: v0.16.0
- Target Device: STM32
- Other extensions you installed (and if the issue persists after disabling them): Keil Pack

When pressing Clean Build button, it doesn't seem like clean really happens:
- is cprj temp file removed ?  If not, could this be considered ?
- same for cmake files 
As of now, when we do Build Clean it seems like nothing is rebuilt

mcgordonite, 2023-03-08:
Thanks @LMESTM, I am able to reproduce this issue. The clean build button isn't correctly passing the `--rebuild` option to cbuild. We'll get a fix out for this.

> is cprj temp file removed ? If not, could this be considered ?

The CPRJ isn't removed by `cbuild --rebuild`. However, from toolbox 1.5.0 `cbuild` can run a build directly on a csolution without needing the intermediate CPRJ. We plan to update the extension to do this, so there would be no CPRJ to clean.

> same for cmake files

This should be handled by correctly passing the `--rebuild` option to the underlying tools.

LMESTM, 2023-03-08:
@mcgordonite  thanks for confirmation and plan for correction - sounds good to me :-)

mcgordonite, 2023-06-12:
Closing: the extension now passes `--rebuild` correctly when running a clean build.

closed 2023-06-12 by mcgordonite