ARM-software/vscode-cmsis-csolution #23: Active solution fail to load the project details

LMESTM, 2023-03-10:
Type: Bug Report

Active solution fail to load the project details. 
A colleague of mine installed the extension 
csolution is listed but not Target Type / Build Type / Project.
The same project works fine on other colleagues environment

[image]

It comes together with an error popup:
[image]

There is also an error that we don't understand in the output channel 
[image]
maybe that can help explaining the root cause ?
 
**Describe the bug**
- OS and Version: windows
- VS Code Version: tbc
- Extension Version: Keil Studio pack for web v1.3.2
- Target Device: STM32
- Other extensions you installed (and if the issue persists after disabling them):

**To Reproduce**
Not sure how to reproduce this issue ...
If there is anything we can look at, just let us know

mcgordonite, 2023-03-13:
Are you working behind a proxy? We've had a few reports of problems, but they seem to be resolved by configuring the environment variables here: "2. Configure an HTTP proxy" in https://github.com/ARM-software/vscode-cmsis-csolution#install-the-cmsis-toolbox.

LMESTM, 2023-03-14:
@mcgordonite - thanks for the hint !
we confirm that the problem is related to PROXY settings and mor particularly from not setting NO_PROXY variable.  Maybe the displayed error might be more explicit for users  ?

mcgordonite, 2023-03-14:
Great, glad to hear the issue was resolved.

> Maybe the displayed error might be more explicit for users ?

Good suggestion. We'll investigate showing a notification or other message with guidance in situations like this.

closed 2023-03-14 by mcgordonite