ARM-software/vscode-cmsis-csolution #24: PROMBLEMS view cannot show 'missing pack' message

UniverGao, 2023-03-28:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 18.04 WSL2 in windows 11 Professional 22H2
- VS Code Version: 1.76.2(user setup) by using 'Remote Development' extention 
- Extension Version:0.17.0
- Target Device:Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz   2.60 GHz
- The 'PROBLEMS' tag didn't show missing pack after executing 'Convert' command by using 'Hello.csolution.yml' but 'OUTPUT' tag showed 'missing pack' message.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone example in 'https://github.com/Open-CMSIS-Pack/csolution-examples.git'
2. cd ./csolution-examples/Hello and execute *code .* command
3. Project convert automatically and *OUTPUT* tag shows missing pack problems
4. Open Hello.csolution.yml but *PROBLEM* tag doesn't show missing pack message

**Expected behavior**
- If one or several packs are missing, errors display in the PROBLEMS view (mentioned in README.md of 'vscode-cmsis-csolution' repository).

**Code sample and logs**
//

**Screenshots**
[image]

**Additional context**
//

mcgordonite, 2023-03-28:
Hi @UniverGao, it's not immediately obvious what the cause is here, I wasn't able to reproduce it with the csolution-examples repository. Some things to check:

1. Has the context view loaded correctly, showing the active project, target & build type? If it hasn't, the extension might not have activated correctly. It should show something like this:
    [image]
2. Is the CMSIS pack root configured correctly in WSL? It's possible the csolution convert command is using a different pack cache from the VS Code extension. Do you have a `CMSIS_PACK_ROOT` environment variable set in WSL? Is the `cmsis-csolution.packCachePath` VS Code setting configured for your WSL environment? I've not tried this in WSL, but I think the default would be having `CMSIS_PACK_ROOT` set to something like `/home/ug/.cache/arm/packs`, and leave the VS Code setting blank.

UniverGao, 2023-03-29:
@mcgordonite 
1. It shows like this but no *ACTION* tag
[image]
2. I redefined the environment variable followed the instruction, 
the variable *CMSIS_PACK_ROOT * is '/home/ug/downloads/cmsis-pack' 
*cmsis-csolution.packCachePath* is set as default and I didn't change it
[image]

best regards

mcgordonite, 2023-03-31:
Hi, one more thing to check: are the packs defined in the cproject or csolution file? Support for packs in cproject files was only added to the spec recently and hasn't been implemented in this extension yet.

UniverGao, 2023-04-07:
> Hi, one more thing to check: are the packs defined in the cproject or csolution file? Support for packs in cproject files was only added to the spec recently and hasn't been implemented in this extension yet.

[image]

You can see that all packs can be installed by using 'cpackget' command

closed 2023-05-07 by UniverGao