ARM-software/vscode-cmsis-csolution #26: Issue with cbuild from vscode

fred-r-perso, 2023-03-28:
I work with this vscode configuration:
https://github.com/fred-r-perso/lwfsm/tree/develop/.vscode

Inspired from:
https://github.com/Open-CMSIS-Pack/vscode-get-started/blob/main/.vscode/tasks.json

I have my project defined here:
https://github.com/fred-r-perso/lwfsm/tree/develop/examples/stm32/cmsis_lwfsm_demo

But the compilation fails:
[image]

The cbuild location is correct:
```
fredr2010@penguin:~/gh/lwfsm$ /home/fredr2010/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/artifacts/vcpkg-artifacts-cmsis/tools.open.cmsis.pack.ctools/1.5.0/bin/cbuild -V
cbuild version 1.5.0 (C) 2023 Arm Ltd. and Contributors
```

So I guess the issue is the fact that the .cprj file does not exist.

Indeed in CLI mode I do not pass the cprj as the parameter:
  ```
  fredr2010@penguin:~/gh/lwfsm/examples/stm32/cmsis_lwfsm_demo$ cbuild lwfsm_demo.csolution.yml --packs --configuration .debug+B-U585I-IOT02A
  ```

But passing the cprj is allowed:
```
cbuild: Build Invocation 1.5.0 (C) 2023 Arm Ltd. and Contributors

Usage:
  cbuild [command] <project.cprj | csolution.yml> [flags]
```

Can you please indicate what is wrong in my setup ?

mcgordonite, 2023-03-29:
Hi @fred-r-perso, thanks for testing this out. The csolution extension currently uses the generated CPRJ file as input to `cbuild`. We plan to update this in future to pass the csolution file as in your CLI command above, but that is a new feature in cbuild we don't support yet.

To generate the CPRJ, you can use the `CMSIS: Convert` command to convert the active csolution to CPRJ, then the build button / command should work. By default the extension also runs this command every time you make a change to the csolution file.

fred-r-perso, 2023-03-29:
Thanks, indeed by providing the cprj the build works fine.
When do you plan nto support the csolution as input ?

mcgordonite, 2023-03-31:
That should be done in a month or so. I'll leave this issue open until it is released.

mcgordonite, 2023-07-10:
Closing: the csolution is now treated as input to the build process.

closed 2023-07-10 by mcgordonite