ARM-software/vscode-cmsis-csolution #27: Cannot update manifest files

Frisc98, 2023-04-13:
Type: Bug Report

**Describe the bug**
- OS and Version: Linux Mint 21
- VS Code Version: 1.77.3
- Extension Version: 1.3.4
No packs are installed but no missing packs are displayed in PROBLEMS view. Running the command "CMSIS: Install required packs for active solution" from palette results in the following error: 

9 FAILED_PRECONDITION cannot update manifest files
ErrorMessage: unable to read Index file /home/user/Documents/keil/cmsis_packs/.Manifest/index.pidx
ErrorMessage: open /home/user/Documents/keil/cmsis_packs/.Manifest/index.pidx
ErrorMessage: no such file or directory

**Screenshots**

[image]

**Additional context**

cmsis_pack folder after initialization doesn't contain .Manifest folder, only .Download, .Web and .Local

mcgordonite, 2023-04-14:
Hi @Frisc98, thanks for reporting this. To help debug, can you let me know how you initialised the pack cache? For example, did you use `cpackget init` or Keil MDK?

Frisc98, 2023-04-14:
I've used this command from cpackget readme: 
`$ cpackget init --pack-root path/to/new/pack-root https://www.keil.com/pack/index.pidx`

I've set pack-root folder as empty folder in Documents and after init it contained  .Download, .Web and .Local

mcgordonite, 2023-04-14:
Thanks, I'm able to reproduce this. We think the problem is that the extension can't handle the readonly file permissions set by `cpackget init`. If you `ls -la` in your pack root you should see the permissions. We'll get that fixed in an upcoming release.

You should be able to work around for now by manually setting the permissions to writable for your user.

Frisc98, 2023-04-14:
Yeah, changing the permissions fixed it. Thank you for your help.

closed 2023-04-14 by Frisc98