ARM-software/vscode-cmsis-csolution #28: Solution service failure: 13 INTERNAL unable to add new project ErrorMessage: project path must be absolute ErrorMessage: invalid GrpcServiceError: 13 INTERNAL unable to add new project ErrorMessage: project path must be absolute ErrorMessage: invalid at Object.create

QGB, 2023-04-14:
```
Solution service failure: 13 INTERNAL unable to add new project ErrorMessage: project path must be absolute ErrorMessage: invalid GrpcServiceError: 13 INTERNAL unable to add new project ErrorMessage: project path must be absolute ErrorMessage: invalid at Object.create (c:\Users\qgb\.vscode-insiders\extensions\arm.cmsis-csolution-0.20.0-win32-x64\dist\desktop\extension.js:2:1276735) at Object.parse (c:\Users\qgb\.vscode-insiders\extensions\arm.cmsis-csolution-0.20.0-win32-x64\dist\desktop\extension.js:2:1277636) at Object.callback (c:\Users\qgb\.vscode-insiders\extensions\arm.cmsis-csolution-0.20.0-win32-x64\dist\desktop\extension.js:2:1275726) at Object.onReceiveStatus (c:\Users\qgb\.vscode-insiders\extensions\arm.cmsis-csolution-0.20.0-win32-x64\dist\desktop\extension.js:2:366988) at Object.onReceiveStatus (c:\Users\qgb\.vscode-insiders\extensions\arm.cmsis-csolution-0.20.0-win32-x64\dist\desktop\extension.js:2:362905) at Object.onReceiveStatus (c:\Users\qgb\.v..

```

mcgordonite, 2023-04-14:
Hi @QGB, where did you see this error message?

Frisc98, 2023-04-19:
You have to initialize VSCode workspace first, then create new solution, otherwise you get this error.

mcgordonite, 2023-10-30:
This issue is now resolved: you can create a Csolution anywhere on disk without an open workspace. More improvements to this form will be coming over the next few weeks.

closed 2023-10-30 by mcgordonite