ARM-software/vscode-cmsis-csolution #29: Issue importing a project which had scatter file

Frisc98, 2023-04-19:
Type: Bug Report

**Describe the bug**
- OS and Version: Win10
- VS Code Version: 1,77.3
- Extension Version: 0.20.0
After importing a project from Keil uVision5 which previously had _"Use memory Layout from Target _Dialog"_ enabled (which generates *.sct file) and then trying to build from VS Code results in linker error because the scatter file gets included in .cproject.yaml

[image]

**To Reproduce**
Steps to reproduce the behavior:
-Create a new project in Keil uVision and tick _"Use memory Layout from Target _Dialog"_  in linker tab when selecting "Options for Target".
-Compile the project so *.sct file gets generated
- Untick the _"Use memory Layout from Target _Dialog"_
- Import project into VS Code using "Convert MDK project to csolution"
- Try building

**Additional context**
I'm trying to build it using arm-none-eabi version 12.2.1, not using AC5/6

mcgordonite, 2023-04-21:
Hi @Frisc98, we're planning to upgrade the μVision to Csolution import in the next few weeks. This will change how scatter files are processed slightly:
1. The new importer will not reference the scatter file that got generated by μVision
2. The new importer will generate a new scatter file in the project folder <project>_<target-name>.sct based on the memory regions defined in uvprojx.
3. The cproject will reference this newly generated scatter file.

It's worth noting that the import doesn't convert an Arm Compiler project to a GCC project.

Frisc98, 2023-06-12:
Any news on this?

mcgordonite, 2023-06-12:
Scatter file support should be present in the latest release. Let us know if you still have issues.

closed 2023-06-12 by mcgordonite