ARM-software/vscode-cmsis-csolution #30: Problems with HelloWorld solution in tutorial

DavidA2014, 2023-04-21:
In the tutorial, for DualCore\HelloWorld.csolution.yml, I see these problems:

```
required pack: ARM::CMSIS not installed
required pack: NXP::K32L3A60_DFP@15.0.0 not installed
```

The .yml contains:

```
  packs:
    - pack: ARM::CMSIS
    - pack: NXP::K32L3A60_DFP@15.0.0
```

but extension 'Arm CMSIS solution' is installed. How can I fix that error?

I am using the ARM MPS2 development board. Which pack should I install?

Frisc98, 2023-04-25:
Sounds like you need to download pack solutions. Try opening up command pallete and run `CMSIS: Install required packs for active solution`

mcgordonite, 2023-07-10:
I'm closing this issue, @Frisc98's solution should resolve your problem. Let us know if you're still having problems.

closed 2023-07-10 by mcgordonite