ARM-software/vscode-cmsis-csolution #32: Sort "Active solution" by alphabetical order

VVESTM, 2023-05-23:
Hello, 

When using "Keil Studio Pack" and "Arm CMSIS csolution" extensions, it can be quite difficult to find the relevant "Active solution" in CMSIS context menu. Effectively, if we have several csolution.yml and cproject.yml files, the list is not sorted. And from one launch to the other, the order may vary.
 
Is it possible to have an alphabetical sort ?
[image]

Thanks in advance for your help.

VVESTM, 2023-05-25:
I am working on Windows.

With VSCode, I open a directory which contains several sub directories. In each subdirectories, I have a cproject.yml/csolution.yml:
[image]
(one cproject.yml per project and per board)

In VSCode, the view is like this:
[image]

In "Active Solution", the different lines are not sorted. (This is not visible on my screen capture as I only took a subset of my projects)

micque01, 2023-05-31:
Hi @VVESTM, thanks for the additional info. Version 0.25.0 of the csolution extension has just been released, which includes sorting of solutions and projects in the context drop-downs. Hopefully this fixes your issue!

VVESTM, 2023-05-31:
Thanks a lot !!

closed 2023-05-31 by mcgordonite