ARM-software/vscode-cmsis-csolution #33: Extension does not work in devcontainers

jankap, 2023-06-15:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 22.04
- VS Code Version: 10.79.1
- Extension Version: 0.27.0
- Target Device:
- Other extensions you installed (and if the issue persists after disabling them):

I'm running VSCode in a devcontainer basing on Ubuntu. `cbuild` etc. is able to build my project properly. But the VSCode extension seems not to work. It only finds a csolution file, no contexts, no projects (the .cproject.yml exists) are shown:

[image]

SW Components are empty, too:
[image]

But they are installed:
[image]

In general, I'm missing configuration options like the path to the SW components, where to search for project files, where to find the FLEXLM server, and so on.

Is the extension not yet ready to be used in remote container development?

Thanks :)

**Screenshots**

**Additional context**

mcgordonite, 2023-06-23:
Hi @jankap, thanks for reporting this issue.

There is indeed a bug with the extension in devcontainers. If you also have the extension installed on your local machine, VS Code will currently try to use that rather than installing the extension in the devcontainer. Since the workspace files aren't on your local machine, the extension fails to load your solution.

We'll release a fix for this soon, so VS Code will install the extension in your remote container.

> In general, I'm missing configuration options like the path to the SW components, where to search for project files, where to find the FLEXLM server, and so on.

- What do you mean by the "path to the SW components" configuration here? Do you mean how to configure where the CMSIS packs are installed, or where the configuration files for software components are stored?
- We've recently added a "Solution" panel to the CMSIS view. This shows an outline of the solution. It's still very much a work in progress, but it already shows files from your solution's projects.

mcgordonite, 2023-06-27:
The latest release, 0.31.0, should restore support for using the extension in devcontainers.

mcgordonite, 2023-07-10:
Closing this issue since the extension should now support devcontainers.