ARM-software/vscode-cmsis-csolution #34: Support for Microlib

JonTBeckett, 2023-06-28:
Type: Feature Request

As I'm looking to migrate from Keil to these VS Code extensions, one thing of high importance is identical compile, assemble, and link commands after switching. They're getting really close with each new release which is great! The one big one that still seems to be missing is support for the microlib. In Keil it's just a checkbox to enable the microlib but the csolution extensions uVprojx converter doesn't seem add the necessary compile and link options when converting a uVprojx file with this checkbox checked. 

This could already be in the plan but I just wanted to express a desire for it. 

Thanks!

mcgordonite, 2023-06-28:
Hi @JonTBeckett thanks for the feedback. Support for microlib is on the roadmap for µVision project import. Hopefully that will be added in the next few weeks.

mcgordonite, 2023-07-10:
Support for microlib was added in the 1.1.0 release of the csolution extension, so I'm closing this issue. Let us know if you still have problems with the µVision conversion.

JonTBeckett, 2023-07-11:
@mcgordonite I look forward to checking out next chance I get! Thanks for the quick and helpful responses!