ARM-software/vscode-cmsis-csolution #35: "Manage Software Component" view doesn't support local packs

LMESTM, 2023-06-29:
If a csolution project is using a local pack version (with path: ) then the PACK may not be installed in CMSIS PACK ROOT,
nevertheless it is available, so I think it would be great if the "Software components" view could consider local packs view as well when building the list of available components.

[image]

mcgordonite, 2023-07-10:
Hi @LMESTM, use of local packs is currently supported if you run `cpackget add path/to/pack`, then reload VS Code. We've got supporting the `path` field on the roadmap for the csolution extension.

jkrech, 2024-10-15:
While CMSIS-Toolbox reflects packs specified in csolution.yml with `path:` the CMSIS-Core-Tools, the backend for the component view is not yet considering these packs.