ARM-software/vscode-cmsis-csolution #40: 1 missing packs failed to install

Idol-check, 2023-09-06:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows_NT x64 10.0.19045
- VS Code Version: 1.81.1
- Extension Version:Arm CMSIS csolution v1.6.1
- Target Device:STM32F103RC
- Other extensions you installed (and if the issue persists after disabling them): keil studio pack;chinese;koroFileHeader;makefile tools;one dark pro;vscode-icons;yaml;
[image]
[image]
[image]

I have an STM32F103RC project written in MDK5. After using "convert project to csolution", I was prompted to install pack. But I have been unable to install it successfully.

Idol-check, 2023-09-07:
Thank you for your reply！
1.I tried installing packages for STM32F3 and STM32F1, but both failed.
2.The following image is the abnormal information I captured through the developer tool, and the remaining information is all the information during the installation process
[image]
all information：
`[Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=info msg="Download and install succeeded" ActualURL="https://sadevicepacksdqaus.blob.core.windows.net/idxfile/index.pidx" ContentLength=131807 ContentType=text/plain DestinationPath="C:\\Users\\chiweiyan\\AppData\\Local\\arm\\packs\\.Manifest\\index.pidx" HttpStatusCode=200 RequestURL="https://www.keil.com/pack/index.pidx"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Geehy APM32F00x_DFP 1.0.0}" readPackID="{Geehy APM32F00x_DFP 1.0.3}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Geehy APM32F1xx_DFP 1.0.0}" readPackID="{Geehy APM32F1xx_DFP 1.0.7}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{APEXMIC APM32F1xx_DFP 1.0.0}" readPackID="{APEXMIC APM32F1xx_DFP 1.0.4}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{GorgonMeducer perf_counter 2.2.1}" readPackID="{GorgonMeducer perf_counter 2.2.2-dev}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Maxim MAX32675 1.1.0}" readPackID="{  }"
time="2023-09-07T22:16:06+08:00" level=warning msg="Load pdsc failed." error="unable to unmarshal Index file C:\\Users\\chiweiyan\\AppData\\Local\\arm\\packs\\.Manifest\\Maxim.MAX32675.1.1.0.pdsc: expected element type <package> but have <html>" name=MAX32675 path="C:\\Users\\chiweiyan\\AppData\\Local\\arm\\packs\\.Manifest\\Maxim.MAX32675.1.1.0.pdsc" vendor=Maxim version=1.1.0

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Geehy APM32S1xx_DFP 1.0.0}" readPackID="{Geehy APM32S1xx_DFP 1.0.1}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Geehy APM32F0xx_DFP 1.0.0}" readPackID="{Geehy APM32F0xx_DFP 1.0.6}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{HDSC HC32F_M14 1.0.0}" readPackID="{HDSC HC32F_M14 1.0.3}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:16:06+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Keil Kinetis_SDK_DFP 2.3.1}" readPackID="{Keil Kinetis_SDK_DFP 2.3.0}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:19:55+08:00" level=error msg="Unable to write downloaded file to disk" ActualURL="https://keilpack.azureedge.net/pack/Keil.STM32F3xx_DFP.2.2.2.pack" ContentLength=102421911 ContentType=application/octet-stream DestinationPath="C:\\Users\\chiweiyan\\AppData\\Local\\arm\\packs\\Keil\\STM32F3xx_DFP\\2.2.2" HttpStatusCode=200 RequestURL="http://www.keil.com/pack/Keil.STM32F3xx_DFP.2.2.2.pack" err="read tcp 192.168.3.48:52363->117.18.232.200:443: wsarecv: An existing connection was forcibly closed by the remote host."

console.ts:137 [Extension Host] Converting solution to CPRJ c:\Users\chiweiyan\Desktop\test\test.csolution.yaml
log.ts:441   ERR [Extension Host] csolution process failed to start
spawn csolution ENOENT
console.ts:137 [Extension Host] csolution process failed to start
spawn csolution ENOENT
p @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
N @ rpcProtocol.ts:455
M @ rpcProtocol.ts:440
H @ rpcProtocol.ts:370
G @ rpcProtocol.ts:296
(anonymous) @ rpcProtocol.ts:161
w @ event.ts:1089
fire @ event.ts:1120
fire @ ipc.net.ts:650
ne.onmessage @ localProcessExtensionHost.ts:583
console.ts:137 [Extension Host] Conversion unsuccessful c:\Users\chiweiyan\Desktop\test\test.csolution.yaml Error: spawn csolution ENOENT
console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:19:55+08:00" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=InstallMissingPacks grpc.service=solutions.Solutions grpc.start_time="2023-09-07T22:16:03+08:00" grpc.time_ms=231709.23 span.kind=server system=grpc

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:19:55+08:00" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=ListProjects grpc.service=solutions.Solutions grpc.start_time="2023-09-07T22:19:55+08:00" grpc.time_ms=0 span.kind=server system=grpc

console.ts:137 [Extension Host] No webview with type cmsis-csolution.context is registered.
p @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
N @ rpcProtocol.ts:455
M @ rpcProtocol.ts:440
H @ rpcProtocol.ts:370
[... cut ...]
console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:19:55+08:00" level=error msg="The pack ID in the pdsc file does not match the path and filename." loadRequestPackID="{Geehy APM32F00x_DFP 1.0.0}" readPackID="{Geehy APM32F00x_DFP 1.0.3}"

console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:19:56+08:00" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=ValidateSolution grpc.service=solutions.Solutions grpc.start_time="2023-09-07T22:19:55+08:00" grpc.time_ms=721.321 span.kind=server system=grpc

console.ts:137 [Extension Host] Event coreTools.validateSolution took 0.726s`

Idol-check, 2023-09-07:
other msg:
[image]
Unable to load schema from 'https://raw.githubusercontent.com/Open-CMSIS-Pack/devtools/schemas/projmgr/2.1.0/tools/projmgr/schemas/csolution.schema.json': Request vscode/content failed unexpectedly without providing any details.

Idol-check, 2023-09-07:
other image:
[image]
[image]
[image]
That's all the information

mcgordonite, 2023-09-08:
Thanks for all of the detail there. This looks like this line is the culprit:
> console.ts:137 [Extension Host] [Core Tools] time="2023-09-07T22:19:55+08:00" level=error msg="Unable to write downloaded file to disk" [...]

One thing to try: can you download the pack using the `cpackget` tool from the command line?
```bash
cpackget add Keil::STM32F1xx_DFP@2.4.1
```
`cpackget` is available as part of [CMSIS Toolbox](https://github.com/Open-CMSIS-Pack/cmsis-toolbox). It will be available in the terminal in VS Code if you have the Environment Manager extension installed and have a vcpkg-configuration.json file at the root of the workspace, or you can install it using the instructions in the Toolbox repository.

Idol-check, 2023-09-08:
Thanks. I used your method and successfully installed the package through the command
‘’‘
cpackget add Keil::STM32F1xx_DFP@2.4.1
’‘’
Now I can also compile my project successfully. 
The remaining issue I have is that I am unable to debug and run. I'm trying to see if I can solve this problem myself.
Thank you very much for helping me solve the above problem.
I also found a faster way to install packages. After downloading the corresponding packages from the website( https://www.keil.arm.com/devices/) myself, I placed them under "C:\Users\xxxxx\AppData\Local\arm\packages\.Download", and then entered the above command to add them more quickly.

mcgordonite, 2023-09-08:
Glad you were able to install the packs 👍 I'll close this issue then. If you have problems with debug and run, you could open an issue on https://github.com/ARM-software/vscode-embedded-debug.

closed 2023-09-08 by mcgordonite

Idol-check, 2023-09-08:
OK  thanks