ARM-software/vscode-cmsis-csolution #41: uv2csolution uses C/C++ defines for the assembler

timrid, 2023-10-19:
Type: Bug Report

**Describe the bug**
- OS and Version:  Windows 11
- VS Code Version: 1.83.1
- Extension Version: Arm CMSIS csolution v1.11.0

I want to convert a uVision .uvprojx project to a csolution project with uv2csolution. But with the following settings in uVision

[image]

I get the following error message when compiling the csolution project:

```
[1/4] Building AS_ARM object CMakeFiles/TestProjekt.dir/C_/SOME_PATH/Test/RTE/Device/STM32L486JGYx/startup_stm32l486xx.o FAILED: CMakeFiles/TestProjekt.dir/C_/SOME_PATH/Test/RTE/Device/STM32L486JGYx/startup_stm32l486xx.o c:\SOME_OTHER_PATH\.vcpkg\artifacts\fd3d78d7\compilers.arm.armclang\6.20.0\bin\armclang.exe -Ic:/SOME_PATH/Test/RTE/_Target_1 -IC:/Users/USER/AppData/Local/Arm/packs/ARM/CMSIS/5.9.0/CMSIS/Core/Include -IC:/Users/USER/AppData/Local/Arm/packs/Keil/STM32L4xx_DFP/2.6.2/Drivers/CMSIS/Device/ST/STM32L4xx/Include -mcpu=Cortex-M4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard  -Wa,armasm,--pd,"STM32L486xx SETA 1" -Wa,armasm,--pd,"_RTE_ SETA 1" -Wa,armasm,--pd,"CONFIG_PATH SETS config/Config.hpp"  -g -O0   --target=arm-arm-none-eabi -c -masm=auto -o CMakeFiles/TestProjekt.dir/C_/SOME_PATH/Test/RTE/Device/STM32L486JGYx/startup_stm32l486xx.o C:/SOME_PATH/Test/RTE/Device/STM32L486JGYx/startup_stm32l486xx.s
Warning: A1950W: The legacy armasm assembler is deprecated. Consider using the armclang integrated assembler instead.
"<command line>", line 3 (column 18): Error: A1516E: Bad symbol 'config', not defined or external
    3 00000000 CONFIG_PATH SETS config/Config.hpp
                                ^
```

This error message when compiling the assembler file surprised me, because `CONFIG_PATH=config/Config.hpp` is actually only set for the C++ compiler and not for the assembler. So I took a closer look at the generated .cproject.yaml and the CONFIG_PATH shows up under the `define` section.

```
project:
  packs:
    - pack: Keil::STM32L4xx_DFP
    - pack: ARM::CMSIS
    - pack: ARM::CMSIS-NN
    - pack: ARM::CMSIS-DSP
  setups:
    - setup: Options for Target_1
      processor:
        trustzone: non-secure
      debug: "on"
      define:
        - CONFIG_PATH: config/Config.hpp               <--------------------
      misc:
        - C:
            - -std=c90
          C-CPP:
            - -fno-rtti
            - -funsigned-char
            - -fshort-enums
            - -fshort-wchar
            - -ffunction-sections
            - -Weverything
            - -Wno-packed
            - -Wno-reserved-id-macro
            - -Wno-unused-macros
            - -Wno-documentation-unknown-command
            - -Wno-documentation
            - -Wno-license-management
            - -Wno-parentheses-equality
          CPP:
            - -xc++
            - -std=c++98
          ASM:
            - -masm=auto
          Link:
            - --map
            - --load_addr_map_info
            - --xref
            - --callgraph
            - --symbols
            - --info sizes
            - --info totals
            - --info unused
            - --info veneers
            - --list
            - --strict
            - --summary_stderr
            - --info summarysizes
  linker:
```

But according to https://github.com/Open-CMSIS-Pack/devtools/issues/607 everything listed under `define` is valid for C/C++ and the assembler. As a workaround it says that you have to use `misc` section. So if I move this manually from `define` to `misc` section the code compiles without any error.

```
project:
  packs:
    - pack: Keil::STM32L4xx_DFP
    - pack: ARM::CMSIS
    - pack: ARM::CMSIS-NN
    - pack: ARM::CMSIS-DSP
  setups:
    - setup: Options for Target_1
      processor:
        trustzone: non-secure
      debug: "on"
      misc:
        - C:
            - -std=c90
          C-CPP:
            - -fno-rtti
            - -funsigned-char
            - -fshort-enums
            - -fshort-wchar
            - -ffunction-sections
            - -Weverything
            - -Wno-packed
            - -Wno-reserved-id-macro
            - -Wno-unused-macros
            - -Wno-documentation-unknown-command
            - -Wno-documentation
            - -Wno-license-management
            - -Wno-parentheses-equality
            - -DCONFIG_PATH=config/Config.hpp            <--------------------
          CPP:
            - -xc++
            - -std=c++98
          ASM:
            - -masm=auto
          Link:
            - --map
            - --load_addr_map_info
            - --xref
            - --callgraph
            - --symbols
            - --info sizes
            - --info totals
            - --info unused
            - --info veneers
            - --list
            - --strict
            - --summary_stderr
            - --info summarysizes
  linker:
    - script: TestProjekt_Target_1.sct
  groups:
    - group: Source Group 1
      files:
        - file: main.cpp
  components:
    - component: ARM::CMSIS:CORE
    - component: Keil::Device:Startup
```

So until the `define-asm` section from https://github.com/Open-CMSIS-Pack/devtools/issues/607 is implemented, uv2csolution should put the defines into the `misc` and not to the `define` section.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/timrid/UvisionTest
2. Open the folder in VSCode
3. Right click `TestProject.uvprojx` and click `Convert µVision project to csolution`
4. Go to `CMSIS` and then click `Build`
5. See error in log

jkrech, 2023-10-24:
Sorry for the inconvenience caused. I am glad to hear you found a way how to fix the problem in your converted project. We have a developer looking into solving this in the converter.

jkrech, 2024-10-15:
This should be fixed when using CMSIS-Toolbox 2.5.0 or higher in combination with uv2csolution 1.5.0

closed 2024-10-15 by jkrech