ARM-software/vscode-cmsis-csolution #43: Software Components view broken in v.1.11.1

AkiMakkonen-AlifSemi, 2023-10-30:
Software Components view shows an error with text
"Unable to find a device matching vendor=Alif Semiconductor:165 name=AE722F80F55D5AS"

If I downgrade to v.1.11.0 (nothing else changed in the environment/project):
v.1.11.0 shows the enabled components 

The pack file was installed locally with cpackget.

mcgordonite, 2023-10-30:
To confirm, is the device `AE722F80F55D5AS` in an unpublished version of the `AlifSemiconductor::Ensemble` pack? I don't see a device entry for it in the latest version (0.5.2).

Also, can you share how the hardware is selected in your csolution / cproject files?

AkiMakkonen-AlifSemi, 2023-10-31:
Hi,
I installed the pack in question from VS Code terminal:

`cpackget add https://github.com/alifsemi/alif_ensemble-cmsis-dfp/releases/download/v0.9.6/AlifSemiconductor.Ensemble.0.9.6.pack`

Targets are defined like this (I have tried with and without the manufacturer part in the device):
  target-types:
    - type: HE
      device: AE722F80F55D5AS:M55_HE
      define:
        - "CORE_M55_HE"
    - type: HP
      device: AE722F80F55D5AS:M55_HP
      define:
        - "CORE_M55_HP"
      

If I downgrade to v.1.11.0 it shows the components.
But...
If I choose the Ensemble pack 0.5.2 and set device AE722F80F55D5AE:M55_HE 
then it seems to work with v1.11.0 and with the latest versions.

mcgordonite, 2023-11-01:
This should be resolved in release 1.11.3. You may need to reference the pack in your csolution / cproject file:
```yaml
  packs:
    - pack: AlifSemiconductor::Ensemble@0.9.6
```

AkiMakkonen-AlifSemi, 2023-11-01:
Yes, all good now. Thanks!

closed 2023-11-01 by mcgordonite