ARM-software/vscode-cmsis-csolution #44: command for getting .bin output file

AkiMakkonen-AlifSemi, 2023-10-30:
Type: Feature Request

command:cmsis-csolution.getBinaryFile returns the .elf or similar format.

It would be useful to have a command for .bin output as well.

Note:
Arm Embedded Debugger extension has the following commands:
command:embedded-debug.getApplicationFile for .elf
command:embedded-debug.getBinaryFile for .bin

mcgordonite, 2023-11-01:
Good suggestion, thanks! We'll add it to our roadmap.

jkrech, 2024-10-15:
cmsis-csolution.getBinaryFiles (solution)
cmsis-csolution.getBinaryFile (current context)

AkiMakkonen-AlifSemi, 2024-10-15:
@jkrech, what do you mean? Those functions you mentioned return the .elf

ReinhardKeil, 2025-02-03:
IMHO this is complete

closed 2025-02-03 by ReinhardKeil

jkrech, 2025-02-03:
Can be read from: `*.cbuild-run.yml`

```  
output:
    - file: out/HID/STM32U585AIIx/Debug/HID.bin
      type: bin
```