ARM-software/vscode-cmsis-csolution #45: command:cmsis-csolution.getBinaryFile returns path to a non-existing .axf file (In Linux environment)

AkiMakkonen-AlifSemi, 2023-10-31:
This is a similar case than https://github.com/ARM-software/vscode-cmsis-csolution/issues/42

Version 1.11.2 works as expected in Windows.

But in Ubuntu 22.04:
v.1.11.0 getBinaryFile works as expected
v.1.11.1 getBinaryFile returns empty string
v.1.11.2 getBinaryFile returns path to non-existing debug+HE.axf file (building with GCC so it should be .elf by default)

In cproject.yaml I set output like this:
  output:
    base-name: $Project$
    type:
      - elf
      - bin

mcgordonite, 2023-11-01:
Has the cbuild.yml generated correctly for the `.debug+HE` context? The extension uses the `output` node in the cbuild file to find the binary name.

AkiMakkonen-AlifSemi, 2023-11-01:
Yes, it seems to be ok. .axf is not mentioned in it and the filenames are ok. I am using cmsis-toolbox 2.1 if that matters.

output-dirs:
    intdir: tmp/blinky/HE/debug
    outdir: out/blinky/HE/debug
    rtedir: RTE
  output:
    - type: bin
      file: blinky.bin
    - type: elf

closed 2023-11-08 by mcgordonite