ARM-software/vscode-cmsis-csolution #46: getProjectPath and getCprjPath commands return erroneous values

AkiMakkonen-AlifSemi, 2023-11-02:
Test done in Windows.

With v.1.11.3 the commands return relative path, but the path is not relative to work folder (in my case c:\workspace\alif_vscode-template_DEV).
 
getProjectPath returns:
..\workspace\alif_vscode-template_DEV\viewfinder\viewfinder.cproject.yaml

getCprjPath returns:
..\workspace\alif_vscode-template_DEV\viewfinder\viewfinder.debug+HE.cprj

With v1.11.0 the commands do work as expected. getProjectPath returns correct relative path and getCprj path return absolute path.
getProjectPath returns:
viewfinder\viewfinder.cproject.yaml

getCprjPath returns:
c:\workspace\alif_vscode-template_DEV\viewfinder\viewfinder.debug+HE.cprj

getSolutionPath on the other hand works with both of these versions.
getSolutionPath returns:
c:\workspace\alif_vscode-template_DEV\alif.csolution.yaml

mcgordonite, 2023-11-03:
Hi, I can reproduce this problem on Windows, we will get a fix out soon.

Out of interest, what is your use case for the `getCprjPath` method? The Open-CMSIS-Pack project plans to deprecate CPRJ in the coming months.

AkiMakkonen-AlifSemi, 2023-11-06:
Hi, currently we are not using those methods. I was searching for a way to clean a single (selected) project using user tasks. Then I noticed these methods do not return the correct values and decided to report the findings. Basically, https://github.com/ARM-software/vscode-cmsis-csolution/issues/47 would solve our problem without the need for getCprjPath

mcgordonite, 2023-11-07:
Hi, this should be fixed in version 1.11.4.

AkiMakkonen-AlifSemi, 2023-11-08:
Good!
I do not see https://github.com/ARM-software/vscode-cmsis-csolution/issues/45 with version 1.11.4 anymore.
coincidence?

mcgordonite, 2023-11-08:
> I do not see https://github.com/ARM-software/vscode-cmsis-csolution/issues/45 with version 1.11.4 anymore.
coincidence?

Not a coincidence, I believe that should be fixed too but I was only able to reproduce it intermittently. I'll close both issues. Thanks for confirming.

closed 2023-11-08 by mcgordonite