ARM-software/vscode-cmsis-csolution #48: Need a proper getting started guide

redbilledpanda, 2024-01-16:
Hello folks

So I installed this Keil studio pack as was recommended. Now I naively thought that creating a "solution" via this extension will hopefully create a barebones skeleton for me, taking care of creating include paths and generating appropriate headers.

However none of that seems to have happened. What it did was create a bunch of yaml files that installed some packs that were apparently not present. yes it create a bunch of json files too under the `.vscode` folder, but those look like they will get populated at run time. Nevertheless, the include path under `c_cpp_properties.json` seems very generic.

So what is the purpose of specifying the hardware? Shouldn't this extension take care of including appropriate header files (and excluding others)? How am I supposed to use this extension?

Elise-Kaminski, 2024-01-16:
Hello,

It looks like you tried to create a solution from scratch. Is that correct?

It might be easier to start from an example project as explained in this [Getting started](https://developer.arm.com/documentation/108029/0000/Get-started-with-an-example-project) chapter. We used the FRDM-K32L3A6 board in the tutorial, but you can look for an example for your board on our [website](https://www.keil.arm.com/boards/). For some boards, we only provide μVision examples in .uvprojx format at this stage. These can be converted as explained in [Convert a Keil μVision project to a csolution project](https://developer.arm.com/documentation/108029/0000/Arm-CMSIS-csolution-extension/Convert-a-Keil--Vision-project-to-a-csolution-project).

If you still want to create a project from scratch, that's ok. The steps to do that are described in [Create a csolution project](https://developer.arm.com/documentation/108029/0000/Arm-CMSIS-csolution-extension/Create-a-csolution-project). Then, my advice would be to check the [official Open-CMSIS-Pack documentation](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-overview.md#overview-of-operation) to find examples of csolution.yml, cproject.yml, and clayer.yml files and understand what CMSIS-Toolbox does for you behind the scenes. We are working on an MDK v6 Getting Started Guide (not released yet) in which we give an overview of CMSIS-Toolbox. Here's an overview diagram:

[image]

We hope this helps. Please let us know if you have further questions.

Elise-Kaminski, 2024-04-02:
> arm_keil_microcontroller_development_kit_(mdk)_getting_started_guide_109350_v6_01_en (2).pdf

Hello @redbilledpanda,
This is the link to the MDK v6 Getting Started Guide on Arm Developer:
https://developer.arm.com/documentation/109350/latest

Thank you.

closed 2024-04-02 by mcgordonite