ARM-software/vscode-cmsis-csolution #49: [Bug]: Cannot generate RTE file automatically

ShiinaKaze, 2024-01-23:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11 Pro 10.0.22631 Build 22631
- VS Code Version: 1.85.1 (system setup)
- Extension Version: v1.24.0
- Target Device: STM32F103C8
- Other extensions you installed (and if the issue persists after disabling them): Arm Keil Studio Pack
- Bug description:
CMSIS Build Manager output:
```log
*** WARNING M634:
  File 'd:/Develop/Projects/VSCodeProjects/demo/Project/RTE/_Debug_STM32F103C8/RTE_Components.h' was not found, use the option '--update-rte' to generate it

*** ERROR M204:
  Path not found: 'd:/Develop/Projects/VSCodeProjects/demo/Project/RTE/_Debug_STM32F103C8'!
```

**To Reproduce**
Steps to reproduce the behavior:
1.Create new CMSIS Solution
2.Wait for package download
3.See error and warning:
```
*** ERROR M204:
  Path not found: 'd:/Develop/Projects/VSCodeProjects/demo/Project/RTE/_Debug_STM32F103C8'!
```

**Expected behavior**
Generate RTE file automatically.

**Code sample and logs**
log:
```log
cbuild d:\Develop\Projects\VSCodeProjects\demo\demo.csolution.yml -t database --context .Debug+STM32F103C8
spawn cbuild ENOENT
cbuild d:\Develop\Projects\VSCodeProjects\demo\demo.csolution.yml -t database --context .Debug+STM32F103C8
info cbuild: Build Invocation 2.2.1 (C) 2023 Arm Ltd. and Contributors
D:/Develop/Projects/VSCodeProjects/demo/demo.cbuild-idx.yml - info csolution: file generated successfully
D:/Develop/Projects/VSCodeProjects/demo/Project/Project.Debug+STM32F103C8.cbuild.yml - info csolution: file generated successfully
D:/Develop/Projects/VSCodeProjects/demo/demo.cbuild-pack.yml - info csolution: file generated successfully
D:/Develop/Projects/VSCodeProjects/demo/Project/Project.Debug+STM32F103C8.cprj - info csolution: file generated successfully
info cbuild: Processing 1 context(s)
info cbuild: Retrieve build information for context: "Project.Debug+STM32F103C8"
================================================================
info cbuild: (1/1) Building context: "Project.Debug+STM32F103C8"

M650: Command completed successfully.

*** WARNING M634:
  File 'd:/Develop/Projects/VSCodeProjects/demo/Project/RTE/_Debug_STM32F103C8/RTE_Components.h' was not found, use the option '--update-rte' to generate it

*** ERROR M204:
  Path not found: 'd:/Develop/Projects/VSCodeProjects/demo/Project/RTE/_Debug_STM32F103C8'!
error cbuild: error executing 'cbuildgen cmake'
error cbuild: error building 'd:\Develop\Projects\VSCodeProjects\demo\Project\Project.Debug+STM32F103C8.cprj'
error cbuild: exit status 1
Process failed with exit code 4294967295
Process failed with exit code 4294967295
```

**Screenshots**
[image]

**Additional context**
none

mcgordonite, 2024-01-29:
Hi @ShiinaKaze, thanks for reporting this issue. It looks like the Csolution extension isn't generating the RTE files before trying to generate the compile_commands.json files needed for intellisense, causing the errors you see in the output channel. We will work on a fix for this.

You can work around this yourself by:
1. Run the "CMSIS: Convert to CPRJ" command from VS Code's command palette (Help > Show All Commands).
2. Restart VS Code.

mcgordonite, 2024-03-26:
This is now resolved: the extension calls cbuild with the `--update-rte` flag when generating the compile_commands.json.

closed 2024-03-26 by mcgordonite