ARM-software/vscode-cmsis-csolution #50: Add Before and After Build script running option

nmennel, 2024-01-31:
Type: Feature Request

We need the functionality from Keil where you can define scripts (e.g. batch or python) to run before or after the build under "Options for Target" - "User".

jkrech, 2024-01-31:
Thanks for raising this topic.
This feature is currently not implemented in the underlying CMSIS-Toolbox (latest 2.2.1) version which is a pre-requisite for exposing it in the extension.
You find the current draft/thinking here:
https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#execute
Please review and let us know whether we are missing something or whether you like what you see.

nmennel, 2024-02-08:
Thank you for the quick reply. I looked at the draft and if it will be possible to execute a Python or Batch script instead of an .exe program, then it would meet our requirements.

vescovd1, 2024-03-22:
This enhancement gets my vote!

mcgordonite, 2024-03-26:
I'm closing this issue because it is best tracked in the underlying CMSIS-Toolbox.

If you're interested in this topic, please check out the proposal here: https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#execute

See also this extension of the proposal: https://github.com/Open-CMSIS-Pack/devtools/issues/1358

closed 2024-03-26 by mcgordonite