ARM-software/vscode-cmsis-csolution #51: Software Components View does not like parenthesis in Csub tag

gauthiergodart, 2024-02-08:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10
- VS Code Version: 1.86.0
- Extension Version: 1.12.7

When a component is defined into the PDSC with parenthesis in a Csub tag (e.g. Csub="Test (parenthesis)"), a bug occurs when one try to select the corresponding component via the Software Components view in VS Code: the component appears selected for a few seconds, but then, it is automatically unchecked. The dependencies of the component are not signaled to the user, and the user gets the feeling the component could not be selected. But, in the cporject.yml, the component is actually referenced.
In other word, there is a bug in the parser used in background, which fails to handle parenthesis into Csub tags in a pdsc file. This bug also reveals wrong error handling: the user is not informed of the issue, and the project remains in a kind of undefined state, with the component being actually added to the environment, without having its dependencies met, and without the user being aware of it.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a test component into any PDSC. Add a Csub tag, with the value containing a parenthesis
2. Create a project with the corresponding pack
3. Try to select the test component with the Software Components view
4. See that the component gets unselected, whilst being actually added to the cproject.yml file

**Expected behavior**

The component should be actually selectable, and its dependencies should be checked

jkrech, 2024-02-09:
Thanks a lot for reporting your problem with using parenthesis in the "Csub" string. 
While reproducing your scenario the schema check for the <component> element in the pdsc file already triggers an error for me. Parentheses are not allowed in "Csub".
When entering the componentID in a cproject.yml file, the YAML schema check also complaints about the parentheses.
I believe this is the root cause for the problem you are seeing.
The IDE tries to add the componentID to the cproject.yml file but it gets refused as invalid.

gauthiergodart, 2024-02-09:
Thanks for your answer. 
It is kind of weird, because the PDSC schema check never complained with this on our side. What is the version of the schema you have been using?

jkrech, 2024-02-09:
Sorry, my fault. I have used "(" as the first character in `Csub="(custom)"` which is not allowed.
I checked that the CMSIS-Toolbox deals with parenthesis in `Csub` correctly. 
The Software Components View needs to be looked at.

mcgordonite, 2024-02-12:
Hi @gauthiergodart, thanks for reporting this. We've found the issue in the Software Components view and a fix will come in the next release.

gauthiergodart, 2024-02-12:
Hi @mcgordonite, thanks a lot for the update. Feel free to get back to me if I can be of further assistance.

mcgordonite, 2024-02-12:
The fix was just released in version 1.25.0 of the extension.

closed 2024-02-12 by mcgordonite