ARM-software/vscode-cmsis-csolution #52: generated solution for GCC link errors

vescovd1, 2024-02-23:
Using:
Windows 10 64bit
VSC 1.86.2
Arm Keil Studio Pack v1.13.0 (pre-release)

Going through new solution set up:
selected New CMSIS solution
selected Target device  STM32F427VITx
selected Blank Solution
selected Compiler GCC
selected solution name and location etc.
opened solution and all packs appear to load and template project created etc.
when I do clean build I get the following errors which I cannot seem to clear.

M650: Command completed successfully.

M652: Generated file for project build: 'C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/tmp/Project/STM32F427VITx/Debug/CMakeLists.txt'
-- The ASM compiler identification is GNU
-- Found assembler: c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/arm-none-eabi-gcc.exe
-- The C compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/tmp/Project/STM32F427VITx/Debug
[1/4] Building ASM object CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S.obj
[2/4] Building C object CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/main.c.obj
[3/4] Building C object CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/RTE/Device/STM32F427VITx/system_stm32f4xx.c.obj
[4/4] Linking C executable "C:\Temp\CMSIS_ST_EXAMPLE\Blank solution\out\Project\STM32F427VITx\Debug\Project.elf"
FAILED: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/out/Project/STM32F427VITx/Debug/Project.elf gcc_linker_script.ld C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/tmp/Project/STM32F427VITx/Debug/gcc_linker_script.ld 
cmd.exe /C "cmd.exe /C "cd /D "C:\Temp\CMSIS_ST_EXAMPLE\Blank solution\tmp\Project\STM32F427VITx\Debug" && c:\Users\vescovd1\.vcpkg\artifacts\2139c4c6\compilers.arm.arm.none.eabi.gcc\13.2.1\bin\arm-none-eabi-gcc.exe -E -P -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -xc -include "c:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/regions_STM32F427VITx.h" "c:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/gcc_linker_script.ld.src" -o "C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/tmp/Project/STM32F427VITx/Debug/gcc_linker_script.ld" && cd "C:\Temp\CMSIS_ST_EXAMPLE\Blank solution\tmp\Project\STM32F427VITx\Debug"" && c:\Users\vescovd1\.vcpkg\artifacts\2139c4c6\compilers.arm.arm.none.eabi.gcc\13.2.1\bin\arm-none-eabi-gcc.exe -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mlittle-endian -DSTM32F427xx -D_RTE_  -g3 -O0    -std=gnu11 -masm-syntax-unified -fomit-frame-pointer -ffunction-sections -fdata-sections -isystem "c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/lib/gcc/arm-none-eabi/13.2.1/include" -isystem "c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/lib/gcc/arm-none-eabi/13.2.1/include-fixed" -isystem "c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/arm-none-eabi/include" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -T "C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/tmp/Project/STM32F427VITx/Debug/gcc_linker_script.ld" --specs=nano.specs --specs=rdimon.specs -Wl,-Map="c:/Temp/CMSIS_ST_EXAMPLE/Blank solution/out/Project/STM32F427VITx/Debug/Project.elf.map" -Wl,--gc-sections -g3 -O0     CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S.obj CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/RTE/Device/STM32F427VITx/system_stm32f4xx.c.obj CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/main.c.obj -o "C:\Temp\CMSIS_ST_EXAMPLE\Blank solution\out\Project\STM32F427VITx\Debug\Project.elf"   && cd ."
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: C:\Temp\CMSIS_ST_EXAMPLE\Blank solution\out\Project\STM32F427VITx\Debug\Project.elf has a LOAD segment with RWX permissions
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Project.dir/C_/Temp/CMSIS_ST_EXAMPLE/Blank_solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S.obj: in function `LoopFillZerobss':
C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S:61:(.text.Reset_Handler+0x38): undefined reference to `_estack'
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S:64:(.text.Reset_Handler+0x3c): undefined reference to `_sdata'
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S:65:(.text.Reset_Handler+0x40): undefined reference to `_edata'
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S:66:(.text.Reset_Handler+0x44): undefined reference to `_sidata'
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S:81:(.text.Reset_Handler+0x48): undefined reference to `_sbss'
c:/Users/vescovd1/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Temp/CMSIS_ST_EXAMPLE/Blank solution/Project/RTE/Device/STM32F427VITx/startup_stm32f427xx.S:82:(.text.Reset_Handler+0x4c): undefined reference to `_ebss'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error cbuild: error executing 'cmake' build
error cbuild: error building 'c:\Temp\CMSIS_ST_EXAMPLE\Blank solution\Project\Project.Debug+STM32F427VITx.cprj'
error cbuild: exit status 1
Process failed with exit code 4294967295
Process failed with exit code 4294967295

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 
The "Blank solution.solution.csolution.yml" seems to have pulled in the correct Device Family Pack:

# A solution is a collection of related projects that share same base configuration.
solution:
  created-for: CMSIS-Toolbox@2.0.0
  cdefault:
  compiler: GCC

  # List the packs that define the device and/or board.
  packs:
    - pack: Keil::STM32F4xx_DFP@2.17.1
    - pack: ARM::CMSIS

  # List different hardware targets that are used to deploy the solution.
  target-types:
    - type: STM32F427VITx
      device: STMicroelectronics::STM32F427VITx

  # List of different build configurations.
  build-types:
    - type: Debug
      debug: on
      optimize: none

    - type: Release
      debug: off
      optimize: balanced

  # List related projects.
  projects:
    - project: Project/Project.cproject.yml

Is this some kind of linker issue?  What am I missing?

vescovd1, 2024-02-23:
I can see the linker script (gcc_linker_script.ld.src) generated in Project/RTE/Device/STM32F427VITx folder

brondani, 2024-02-23:
@vescovd1 To reference a custom linker script into the solution, add a [`linker`](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#linker) node in the `Project.cproject.yml` file like this:
```
  linker:
    - script: path/to/linkerScript.ld
```

vescovd1, 2024-02-23:
Ok that works ... no link errors under GCC

I also build with compiler: AC6 and CLANG  is there a way to specify only use linker script in context of GCC compiler?
i.e. using the "for-context" tag?

vescovd1, 2024-02-23:
works

closed 2024-02-23 by vescovd1