ARM-software/vscode-cmsis-csolution #53: Example projects not detected from the Create Solution view

gauthiergodart, 2024-02-23:
Type: Bug Report

## Bug description
- OS and Version: Win 10
- VS Code Version: 1.86.2
- Extension Version: 1.26.1

We use a confidential pack. It has been installed with success via cpackget, with an absolute path.
When creating a new solution, and selecting our device, the "Template and Examples" selector only proposes creating a Blank solution, despite the fact that we provide examples in the pack, and the pdsc has been edited so that these should be detected (example tag, with an environment tag set to the type "csolution", and load tag pointing to the csolution file).

```XML
<example name="***" folder="***" doc="****" version="1.0.2">
	<description>****</description>
	<board vendor="****" name="****" />
	<project>
		<environment name="uv" folder="./ARM" load="****.uvprojx" />
		<environment name="csolution" folder="./KeilStudio" load="****.csolution.yml"/>
	</project>
	<attributes>
		...
	</attributes>
</example>
```

[image]

## Expected Behavior
We supposed our examples would appear in the "Template and Examples" dropdown menu. Did we miss something?

Thanks in advance

JoeTheGuitarist, 2024-02-23:
Thanks for your feedback. We don't yet support selecting "local" examples from packs in the Create Solution view, but we have this feature in our backlog.

gauthiergodart, 2024-02-26:
@JoeTheGuitarist thanks for your answer. Glad to know that this feature is already in your backlog 👍

closed 2024-10-15 by jkrech

gauthiergodart, 2024-10-15:
Thank you for the information.
I'll have a look a that 👍