ARM-software/vscode-cmsis-csolution #55: Can't create solution with Type Error at 'extension.js' of the 'Arm CMSIS SOLUTION' extension

2447328758, 2024-03-22:
Type: Bug Report

**Can't create solution with Type Error at 'extension.js' of the 'Arm CMSIS SOLUTION' extension**
- OS and Version: windosws 10 family 22h2
- VS Code Version: 1.77.1
- Extension Version: 1.33.0
- Target Device: stm32f103c6
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.
THE WARNING MESSAGE IS FOLLOWING :

`Solution service failure: e.listPackages is not a function
TypeError: e.listPackages is not a function
    at e:\Program Files\VSCode-win32-x64-1.77.1\data\extensions\arm.cmsis-csolution-1.33.0-win32-x64\dist\desktop\extension.js:37:203118
    at Generator.next (<anonymous>)
    at e:\Program Files\VSCode-win32-x64-1.77.1\data\extensions\arm.cmsis-csolution-1.33.0-win32-x64\dist\desktop\extension.js:37:201469
    at new Promise (<anonymous>)
    at s (e:\Program Files\VSCode-win32-x64-1.77.1\data\extensions\arm.cmsis-csolution-1.33.0-win32-x64\dist\desktop\extension.js:37:201214)
    at t.CreateSolutionWebviewMain.configureVcpkgForNewSolution (e:\Program Files\VSCode-win32-x64-1.77.1\data\extensions\arm.cmsis-csolution-1.33.0-win32-x64\dist\desktop\extension.js:37:203060)
    at t.CreateSolutionWebviewMain.<anonymous> (e:\Program Files\VSCode-win32-x64-1.77.1\data\extensions\arm.cmsis-csolution-1.33.0-win32-x64\dist\desktop\extension.js:37:195657)
    at Generator.next (<anonymous>)
    at s (e:\Program Files\VSCode-win32-x64-1.77.1\data\extensions\arm.cmsis-csolution-1.33.0-win32-x64\dist\desktop\extension.js:37:190439)`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'cmsis solution' sidebar
2. Click on 'create a new solution'
3. Select the target device 'stm32f103c6'.
4. Input the project name.
5. Select the complier for Arm complier 6
6. input the solution name and location.
7. click the create button.
8. See error

**Expected behavior**
please help me solve the problem.

**Screenshots**
[image]

mcgordonite, 2024-03-22:
Hi, thanks for reporting this issue. Do you have the [Arm Environment Manager](https://marketplace.visualstudio.com/items?itemName=Arm.environment-manager) extension installed too? If so, which version?

D4box, 2024-03-23:
I have the same quesant, the version of Arm Environment Manager is v1.0.5.

mcgordonite, 2024-03-25:
Hi @D4box, does updating to the latest Environment Manager release fix this for you?

2447328758, 2024-03-29:
> Hi, thanks for reporting this issue. Do you have the [Arm Environment Manager](https://marketplace.visualstudio.com/items?itemName=Arm.environment-manager) extension installed too? If so, which version?

Yes,and it is v1.0.5
[image]

2447328758, 2024-03-30:
I don't remember  the every step I just did. But it works fine now! Maybe for the reinstalling of 'Arm Debbuger' extenssion which depends on the 'Arm Environment Manager' extension. But I just see the version number of 'Arm Environment Manager' is still '1.0.5' in my vscode1.77.1. It seems that it is the same with the inital environment. And I truly didn't konw what happened.
Besides, I firstly think about the incompatibility of the vscode version and the extension version. So I installed the lastest version of vscode and it worked correctly. And I noticed that the lastest version of 'Arm Environment' in the lastest vscode is '1.1.10' but the '1.0.5' in the '1.77.1' vscode.
Hope somebody will find the true cause. My hands are tied now.

mcgordonite, 2024-04-02:
Thanks @2447328758. I investigated the release history for the Environment Manager extension. The minimum VS Code version was increased to `1.82.0` in version `1.0.6` of the Environment Manager. That means you were stuck on an old Environment Manager until you updated VS Code.

Version `1.0.5` was not compatible with later versions of the Csolution extension, leading to this problem. We will improve handling of issues like this for future updates.

I'll close this issue. @D4box let us know if the same steps do not fix your problem.

closed 2024-04-02 by mcgordonite