ARM-software/vscode-cmsis-csolution #56: modifying project yml aggressively drops project from solution context

vescovd1, 2024-03-22:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10
- VS Code Version: 1.87.2
- Extension Version: ARM CMSIS solution v1.33.0, Keil Solution Pack 1.18.1
- Target Device: STM32xx
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.  Solution with multiple projects, modify (editing) one projects yml (for example simply just adding a new file to the build list) drops the project from the active solutions context. This is extremely annoying as it seems every time I make even minor changes to the project yml the project is dropped from the solution context and if I try to trigger a build it builds some other project in the solution instead of the project I modified the yml for. The only solution I have found is shut down VSC and restart, then the modified project reappears in the solution's context but is unselected.   

**To Reproduce**
Steps to reproduce the behavior: See above.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Code sample and logs**
- Code sample
- `tasks.json`

**Screenshots**

**Additional context**
 It appears the yml syntax scanner is too aggressive and drops projects from solution's context and does not allow simple yml edits.

mcgordonite, 2024-03-22:
Thanks for reporting this issue, it is very annoying. We'll get a fix out for it as soon as possible.

mcgordonite, 2024-03-22:
Version 1.33.1 released today has an initial fix for this. The projects will not be dropped from the context if the YAML is invalid.

closed 2024-03-26 by mcgordonite