ARM-software/vscode-cmsis-csolution #58: Csolution not loading context on VSCode startup: need to disable-reload-enable extension to work properly

siltau, 2024-03-26:
Type: Bug Report

**Describe the bug**
- OS and Version: Linux 6.5.0-26-generic #26~22.04.1-Ubuntu x86_64 x86_64 x86_64 GNU/Linux
- VS Code Version: 
                      Version: 1.87.2
                      Commit: 863d2581ecda6849923a2118d93a088b0745d9d6
                      Date: 2024-03-08T15:14:59.643Z
                      Electron: 27.3.2
                      ElectronBuildId: 26836302
                      Chromium: 118.0.5993.159
                      Node.js: 18.17.1
                      V8: 11.8.172.18-electron.0
                      OS: Linux x64 6.5.0-26-generic snap
- Extension Version:v1.33.2
- Target Device:
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is: at every startup of VSCode the CMSIS-csolution extension does not load the context properly. In order to make the extension work properly you must disable it, hit the "Reload required" button, then enable it again. 
Note: the reload command "@command:workbench.action.reloadWindow" doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Open VSCode
2. Click on CMSIS Solution extension tab
3. Solution is not loaded properly
4. See error

**Expected behavior**
At the startup the CMSIS Extension load the solution\s it finds in the workspace.

**Code sample and logs**
N.A.

**Screenshots**
[image]
[image]
[image]
[image]

**Additional context**

mcgordonite, 2024-03-27:
Hi @siltau, thanks for reporting this issue. To help debug, please send the logs from the developer tools panel in VS Code.

To do this:
1. Run the `Developer: Toggle Developer Tools` command from the command palette (CTRL-SHIFT-P).
2. Open the Console tab in the developer tools.
3. Right click anywhere in the Console tab and select `Save As…`.

siltau, 2024-03-27:
Hi @mcgordonite, here the log requested:

[after-disable-reload-enable-vscode-app-1711548637429.log](https://github.com/ARM-software/vscode-cmsis-csolution/files/14774588/after-disable-reload-enable-vscode-app-1711548637429.log)
[opening-vscode-app-1711548637429.log](https://github.com/ARM-software/vscode-cmsis-csolution/files/14774589/opening-vscode-app-1711548637429.log)

Hope that helps.

mcgordonite, 2024-03-28:
Thanks for sending the logs. It looks like the Csolution extension is activating successfully, but something is either going wrong with listing the solutions or loading the active one.

Some more things to try:
* After starting VS Code (without the disable-reload-enable steps), rename the csolution file to change the extension (e.g., `My.csolution.yamlx`). Then change it back. Does the view load?
* Uninstall all other extensions (including the Arm extensions) and try loading the same folder in VS Code. You could create a fresh profile instead if this is easier: https://code.visualstudio.com/docs/editor/profiles.
* Check the setting `cmsis-csolution.exclude` is not configured in the user or workspace settings.json file.

siltau, 2024-03-29:
Hi @mcgordonite, following my comments:

> Some more things to try:
> 
>     * After starting VS Code (without the disable-reload-enable steps), rename the csolution file to change the extension (e.g., `My.csolution.yamlx`). Then change it back. Does the view load?
--> This works, after renaming the file as you suggested the context has been loaded properly.

> 
>     * Uninstall all other extensions (including the Arm extensions) and try loading the same folder in VS Code. You could create a fresh profile instead if this is easier: https://code.visualstudio.com/docs/editor/profiles.
-->  Haven't done this yet but I will try asap.

> 
>     * Check the setting `cmsis-csolution.exclude` is not configured in the user or workspace settings.json file.
--> I can confirm that setting is not configured neither for the user or the workspace.

jkrech, 2024-10-15:
@siltau, is there still a problem in recent extension releases? Note version 1.40 was released last week. Please reopen if you still see the problem

closed 2024-10-15 by jkrech