ARM-software/vscode-cmsis-csolution #59: Installing required packs for solution

atoyx, 2024-03-28:
Type: Feature Request

[image]
i need another way to download this package ,because the time we cost is not visiable

jkrech, 2024-03-28:
I am not sure I do understand your reason for needing another way to download the pack Keil::STM32H7xx_DFP@3.1.1 however you can run the pack installer of the CMSIS-Toolbox named `cpackget` from a terminal window in VSCode.
```
  cpackget add Keil::STM32H7xx_DFP@3.1.1
```
You can also locate the require pack version on the web in a browser:
https://www.keil.arm.com/packs/stm32h7xx_dfp-keil/versions/
The "Get Pack" drop down provides you the option to download it on your computer
Afterwards you can point to that file using cpackget:

```
  cpackget add Keil.STM32H7xx_DFP.3.1.1.pack
```

If you already have a pack file download you can use the above command as well.
Please let us  know whether this answers your question.

atoyx, 2024-03-28:
[image]
thanks a lot , i use cpackget load my package. But there still have a problem casuse i have set `cmsis-csolution.packCachePath` as `D:\\Users\\xx\\pack`, extension can't find the installed package. where are the platform-specific default path

atoyx, 2024-03-28:
platform-specific default path is `C:\Users\<your name>\AppData\Local\arm\packs`

atoyx, 2024-03-28:
i found the answer
[image]

closed 2024-03-28 by atoyx