ARM-software/vscode-cmsis-csolution #60: reading gpdsc failed stm32cubemx

atoyx, 2024-04-12:
Type: Bug Report

Description
OS and Version: Win 10
VS Code Version: 1.87.2
Csolution Extension Version: 1.34.1
[image]
when i added the cubemx software module to my project , i got an compilation error

Expected Behavior
fix

jkrech, 2024-04-12:
Note that your problem originates from the fact that the component you have selected: `Keil::Device:STM32Cube Framework:STM32CubeMX` is a so called "generator component". The integration of the STM32CubeMX generator today only works 
a) on Windows
b) in MDK uVision and Arm DS
The use in the context of CMSIS csolution projects and CMSIS-Toolbox is quite restricted and requires many manual steps.
We are currently working on a new generator integration concept which implemented in CMSIS-Toolbox 2.3.0 but the updated STM32 DFPs are in development and will be published over the coming months.
The plan is to completely remove the use of gpdsc file for the STM32 series.

[Use Generators (i.e. CubeMX)](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-tools.md#use-generators-ie-cubemx)

jkrech, 2024-10-15:
We have released new pack versions for STM32 device series last week. These packs are using the new global generator integration for STM32CubeMX.
The CMSIS-Solution extension 1.40 release last week as well supports the launch of the Generator from the CMSIS outline view.

[image]

closed 2024-10-15 by jkrech