ARM-software/vscode-cmsis-csolution #61: CMSIS Support for Texas Instruments TM4C123

bavleym, 2024-04-17:
Type: Feature Request

The TM4C123 does not come with CMSIS debugging traditionally but the platform it is built on has CMSIS natively. As of now in the Tiva family the only device that is recognized by CMSIS when connected to VS Code is the TM4C129 series of the Tiva. BUT, there was a CMSIS header created for the TM4C123 (and other devices) available [here](https://github.com/dannyf00/CMSIS-framework-for-TM4-devices). What I want to know/support/feature for is how can these files be ported into VSCode to be used with the device.

ReinhardKeil, 2024-04-19:
You should put this type of questions to the TI user forum.  TI can publish the CMSIS-Packs to support their device offering. We at Arm don't create that.

I hope this helps.

bavleym, 2024-04-19:
Hi,TI already responded to me and provided the software packs for CMSIS but I want to know how can I use them on VS code? Bavley Mobarak

ReinhardKeil, 2024-04-20:
If this is a local pack, than you should be able to activate this with the *cpackget* command line tool of the CMSIS-Toolbox.

Take a look here https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-tools.md#add-packs

jkrech, 2024-10-15:
No further response from user.
Please reopen if the problem persists.

closed 2024-10-15 by jkrech