ARM-software/vscode-cmsis-csolution #62: Multiproject - Manage Solution - Active Projects

FerdinandEn, 2024-04-19:
Type: Bug Report

**Describe the bug**
OS and Version: Windows 11
VS Code Version: 1.88
CMSIS Solution 1.34.3

I have a multiproject for my student teaching placement. I was able to open and convert the entire project.

Within the project it was not possible to change the active project under Manage Solution. By adjusting the entry in Praktikum.cbuild-set.yml, it was possible to switch. It was then also possible to adjust the targets in Manage Solution.

jkrech, 2024-04-22:
Could you please provide a reproducer. 
If I understand you correctly the initial configuration via the Manage Solution editor was impossible. What was displayed?
After you had edited the file manually you were able to configure the contexts via Manage Solutions.

Note: you can also update the cbuild-set.yml file via the commandline using e.g. `csolution convert Praktikum.csolution.yml -S -c <context1> -c <context2>`

FerdinandEn, 2024-04-22:
I have created and uploaded an example. The project was first created with uVision and then the multi-project was converted to VSC (The screenshot was created after generating).

[Multitest.zip](https://github.com/ARM-software/vscode-cmsis-csolution/files/15058685/Multitest.zip)

[image]

[image]

jkrech, 2024-04-22:
The reason why the project2 is not selectable is because the uv2csolution converter has created two different `target-type` entries for project1 ("Proj1_Target1") and for project2 ("Proj2_Target1") . The Manage Solution is designed to only allow to build a single `target-type` at a time.

At this point you will need to manually edit the csolution.yml file using the same target-type name for both project1 and project2.

generated:
```
    - project: Project1/project1.cproject.yml
      for-context:
        - +Proj1_Target_1
        - +Proj1_Target_2
        - +Proj1_Target_3
    - project: Project2/project2.cproject.yml
      for-context:
        - +Proj2_Target_1
        - +Proj2_Target_2
        - +Proj2_Target_3
```

If you know that both uvprojx files for project1.uvprojx and project2.uvprojx are configured identical you can remove the Prj2_Target_n target-types and share the target-type.
```
    - project: Project1/project1.cproject.yml
    - project: Project2/project2.cproject.yml
```

FerdinandEn, 2024-04-22:
CMSIS Solution 1.34.6

Only one project can be compiled at a time in uVision. Only in batch mode can several projects and targets be processed. To change a project, it must first be activated, then the targets can be selected. The project/target can then be compiled. 

I have just tested the new Csolution version. It is now possible to switch between projects. To do this, you have to change the target under Active Targets and then select the corresponding project under Active projects. I was able to successfully switch between projects and targets. I do not find the solution optimal. For controls, I use the following target identifiers: Debug, Release, Ulink Debug, Ulink Relase, J-Link Debug, ... This makes it difficult to recognize the assignment between target and project.

Translated with www.DeepL.com/Translator (free version)

jkrech, 2024-04-22:
A uVision project target combines the aspects of Hardware selection, Build Options and Debug adapter.
In Keil Studio the solution allows you to configure the Hardware selection in a `target-type`, the build Options like optimization level and generation of debug symbols in `build-type`. 
For each debug adapter you can specify a named configuration in `launch.json` and `tasks.json`.

I assume that you have a single hardware target, which would mean, that there is just a single target-type required.
I do expect that two debug-types `Release` and `Debug` are sufficient.
In case different cprojects require different build flags, these can be set on the project level.

I would expect that this will make it much easier to maintain the common settings of the different projects.

FerdinandEn, 2024-04-23:
Thank you for the clarification

closed 2024-10-15 by jkrech