ARM-software/vscode-cmsis-csolution #63: Multiproject - Error Message for Include

FerdinandEn, 2024-04-19:
Type: Bug Report

**Describe the bug**
OS and Version: Windows 11
VS Code Version: 1.88
CMSIS Solution 1.34.3

Project Structure

CompleteProject
  - project-1
  - project-2
  - ...
  - project-n
  - Common
    - src-1 (C files)
	- src-2 (c files)
	- include (h files)

In uVision the include was set to ..\Common\Include	

I got the error

M650: Command completed successfully.

*** ERROR M204:
  Path not found: '../Common/Include'!
error cbuild: error executing 'cbuildgen cmake'

jkrech, 2024-04-22:
It would be certainly helpful to have a minimal reproducer. From the above description it is not clear to me, where the additional include has been set. I assume that you have a csolution.yml file. If that file is located in the parent folder of all project-n and the Common folder, and you describe the include in the csolution, then the specified relative paths need to be specified relative to the csolution file `Common/Include`
Or are you automatically converting a uvmpw uVision project using uv2csolution?

FerdinandEn, 2024-04-22:
The problem occurred on my home office computer. In my Office computer, I could not find the problem either in my sample project or in the project used on the home office computer. However, I converted the project again and did not use the project with the problem.  For the sake of completeness, here is my example project.

[Multitest.zip](https://github.com/ARM-software/vscode-cmsis-csolution/files/15058786/Multitest.zip)