ARM-software/vscode-cmsis-csolution #65: Debugging Start (STM32)

FerdinandEn, 2024-04-19:
``Type: Feature Request

OS and Version: Windows 11
VS Code Version: 1.88
CMSIS Solution 1.34.3

If I start debugging with the default settings (ARM debugger), no flash process is carried out. To start the flash process you have to execute Run. Can you give me a tip on how to adjust the entry in launch.json so that the device is flashed before the debugger is started.

launch.json
> 
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Arm Debugger",
            "type": "arm-debugger",
            "request": "launch",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${command:arm-debugger.getApplicationFile}",
            // "cmsisPack": "${command:cmsis-csolution.getTargetPack}",
            // "deviceName": "${command:cmsis-csolution.getDeviceName}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "deviceName": "${command:device-manager.getDeviceName}",
            "processorName": "${command:cmsis-csolution.getProcessorName}",
            "probe":"ST-Link",
            "programMode": "auto"
        }
    ]
}

thegecko, 2024-04-19:
In VS Code you can chain tasks together or have a task run before a launch configuration.

In a task configuration the entry is:

```json
"dependsOn": "some-pre-task"
```

and in launch configurations it's:

```json
"preLaunchTask": "my-flash-task"
```

So you can create a flash/run task and set it to run before debug.

Hope this helps!

FerdinandEn, 2024-04-20:
Thanks for your help. I have very little experience with the use of VSCode

I have modified my code 

launch.json

> {
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Arm Debugger",
            "type": "arm-debugger",
            "request": "launch",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${command:arm-debugger.getApplicationFile}",
            // "cmsisPack": "${command:cmsis-csolution.getTargetPack}",
            // "deviceName": "${command:cmsis-csolution.getDeviceName}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "deviceName": "${command:device-manager.getDeviceName}",
            "processorName": "${command:cmsis-csolution.getProcessorName}",
            "preLaunchTask": "Flash Device",
            "probe":"ST-Link",
            "programMode": "auto"
        }
    ]
}

and tasks.json

> {
    "version": "2.0.0",
    "tasks": [
        {
            "type": "arm-debugger.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${command:arm-debugger.getApplicationFile}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "deviceName": "${command:cmsis-csolution.getDeviceName}",
            "processorName": "${command:cmsis-csolution.getProcessorName}",
            "problemMatcher": [],
            "label": "Flash Device",
            "probe": "ST-Link"
        }
    ]
}

Now I got the following message

[image]

I hope someone can tell me what I am doing wrong. I have already tested different spellings for the label.

FerdinandEn, 2024-04-20:
**Addition:**

I have just noticed that the problem only occurs when I start the debugger from the CMSIS view.

If you start the debugger from Run and Debug, everything works, including flashing.

@thegecko Thank you very much for your help

FerdinandEn, 2024-04-20:
`    "version": "0.2.0",
`     "configurations": [
 `        {
 `            "name": "Arm Debugger",
 `          "type": "arm-debugger",`

thegecko, 2024-04-20:
Yes, unfortunately there is a bug in VS Code which prevents these chains from running when executed programmatically:

https://github.com/microsoft/vscode/issues/157041

closed 2024-04-23 by FerdinandEn