ARM-software/vscode-cmsis-csolution #66: Importing existing uVision project ignores "[ ] Include in Target Build"

billev, 2024-05-09:
Type: Bug Report

**Bug Description**
When importing an existing uVision project, components that are marked for non-inclusion in the build (that is, don't have "Include in Target Build" checked) are nonetheless included in the build. 
- Windows 11, up-to-date patches
- VS Code 1.89.0
- Arm CMSIS Solution version v1.34.7
- Target Device STM32F411V
- Keil Studio Pack, C/C++ (Microsoft), clangd (LLVM), Memory Inspector (Eclipse CDT Cloud), Peripheral Inspector (Eclipse CDT Cloud), YAML (Red Hat)

I imported an existing uVision project. Over time we've used different NOR flash chips, support for which had been added to the project, then turned off as no longer needed. I imported that to Arm CMSIS Solution, and those components were imported (ok), and included in the build (not ok), leading to duplicate symbol errors at link time.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a uVision project.
2. Add the "CMSIS Driver" pack.
3. In "Manage Runtime Environment", enable two of the drivers in the "Flash" section, for instance AT45DB641E and AT45DB642D
4. In "Options for component class 'CMSIS Driver', expand the Flash section, select each driver and for each turn OFF "Include in project build"
5. In VS Code use CMSIS to import the uVision project.
6. Build
7. See linker error about duplicate symbols
8. Examine the .cproject.yml file and notice both components listed.

**Expected behavior**
Components marked as "DO NOT Include in target build" should not be included in the component list in the .cproject.yml file.

mguzmanm, 2024-05-31:
Working on this issue.

jkrech, 2024-06-04:
Thanks for reporting the issue. I assume that you were able to manually remove the "disabled" component so you are not blocked.
Just a question, does the uVision project have multiple targets or just one?
If it just has one, could you add a second target without any other modifications and let me know whether the disabling of components is now working in the converted solution?

mguzmanm, 2024-06-26:
Fixed. Node "not-for-context" was added to point out excluded components in .cproject file.

Change will appear in next release uv2csolution 1.5.0.

jkrech, 2024-10-15:
Please, reopen if uv2csolution 1.5.0 does not solve your problem

closed 2024-10-15 by jkrech