ARM-software/vscode-cmsis-csolution #67: Click on "Create new solution" to prompts network error。

haiqinJack, 2024-05-23:
**Describe the bug**
- OS and Version: MacOS 12.6
- VS Code Version: 1.89.1
- Extension Version: v1.18.2
- Click on "Create new solution" to  prompts network error。
﻿
[image]

mcgordonite, 2024-05-28:
Hi @haiqinJack, it looks like a process the extension uses to read CMSIS packs has failed to start. It would help to see the logs. To get the logs:
1. Open VS Code.
2. Run "Developer: Toggle Developer Tools" from the command palette (CMD-SHIFT-P).
3. Open the Console view in the developer tools.
4. Open the Create Solution UI.
5. Right click the text in the Console view and select "Save As".

haiqinJack, 2024-06-01:
@mcgordonite thank.

[image]

mcgordonite, 2024-06-07:
Hi, sorry for the slow response. Thanks for sending over the logs. The `14 UNAVAILABLE` messages in the logs confirm the issue was related to the child process the extension uses to read CMSIS packs, but there aren't any clues as to the cause.

More things to try:
* You are on quite an old version of the Csolution extension. The latest is 1.35.0 (see here: https://marketplace.visualstudio.com/items?itemName=Arm.cmsis-csolution). Does updating help? You may be on an older version of VS Code too.
* Are you working behind a corporate proxy or equivalent?

haiqinJack, 2024-07-01:
@mcgordonite thank.i have created a new system user and can now use it.

closed 2024-07-03 by mcgordonite