ARM-software/vscode-cmsis-csolution #68: STM32H7  Conversion  from uVision to Keil Studio does not work

FerdinandEn, 2024-05-27:
Type: Bug Report

**Describe the bug**
- OS and Version: Win 11
- VS Code Version:
- Keil MDK 5.40 
- Extension Version: 1.34.9 
- Target Device: STM32H753ZIT  Pack 4.0.0-dev0
-  I have created a project with MDK 5.40 with Pack 4.0.0-dev0. I was able to create, use CubeMX, compile and debug the project. I then wanted to convert the project to Keil Studio format. An error occurs both from MDK and from VisualCode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Code sample and logs**
- Code sample
- `tasks.json`

**Screenshots**

**Additional context**

mguzmanm, 2024-05-28:
Issue will be fixed soon.

mguzmanm, 2024-06-04:
Fixed. File extension different from ".gpdsc" will be ignored when retrieving bootstrap components specified in bundle of gpdsc file. Fix should be included in next uv2csolution release 1.5.0

Meanwhile, as workaround please remove <gpdsc> </gpdsc> section from uvptojx.file until 1.5.0 is realeased.

FerdinandEn, 2024-06-04:
I have tested the workaround. After commenting out the >gpdscs< entry, I was able to convert the project. I then tried the following:
- csolution >projekt<.csolution.yml list generators --verbose
- csolution >projekt<.csolution.yml run --generator CubeMX --context >Project<+Target_1

-> CubeMX opens, but a new IOC file is created

The problem relates to the target name. The target name specified is >Target 1<. This identifier contains a space. The uVision creates a directory with the name >Target 1<. The converter creates a new empty directory with >Target_1< and generates the file >project<.cgen.yml there. As the ioc file is not copied, this leads to the creation of a new CubeMX file.

After changing the target name to >Target1<, the project including the ioc file was correctly integrated.

I have noticed another problem with uVision. If you change the name of the target, the directory under /STM32CubeMX/>target name< is not renamed. If you now start CubeMX, a new directory is created with the new target name and a new CubeMX file is created without the previously made entries.

FerdinandEn, 2024-06-05:
I have further investigated the behavior of uVision with the new CubeMX generator (yml). I noticed the following problem:

In the CubeMX version (gpdsc), the CubeMX file is saved under the following path:

`RTE\Device\<STM32....>\STCubeGenerated`

All targets that use the `<STM32....>` device also use the above directory and the generated files stored there.

In the new version of the CubeMX generator (yml), the CubeMX file is located under the directory:

`STM32CubeMX\<Target Name>\STM32CubeMX`

If you change the target and restart CubeMX under RTE, a new CubeMX file is created. If you make changes and generate the code, it is stored under the new target directory.

In the uVison project, it looks like only the generated code of the first target is used, even if you switch to the second target. The CubeMX settings in the second target are therefore useless.

I have not looked any further at how this affects the conversion in Keil Studio.

Is this the right place to report this or should I create a ticket?

jkrech, 2024-06-05:
Thanks a lot for your feedback:
a) For issues with MDK uVision and its project  I suggest to create Arm support tickets, please.
b) I confirm the issue with the spaces in the uVision project >targets< and the "mismatch" of directories after converting to the CMSIS Solution project format. 
Unfortunately the project management concepts of uVision and CMSIS Solution are not matching completely.
As you can see from the global generator here:
https://github.com/Open-CMSIS-Pack/devtools/blob/8546fef2d9ef492bb3ec00c2a7bf7cf057a96769/tools/projmgr/templates/global.generator.yml#L6
each `target-type` has a separate generator directory. This changes the scope of generators significantly.
c) The fact that in uVision the second target is loading the files from the first target is a bug.

closed 2024-10-15 by jkrech