ARM-software/vscode-cmsis-csolution #70: Starting CubeMX via a task

FerdinandEn, 2024-06-13:
Type: Question and Feature Request

Csolution 1.35.0

I am new to the use of VSCode. Therefore, I am not very familiar with the use of `tasks`. To start CubeMX, an instruction with the following structure is required.

`csolution BlinkTest.csolution.yml run --generator CubeMX --context BlinkTest.Debug+STM32H753ZITx`

I have tried to start this instruction via a `task`. I have used the following setting for this:

````
{
    "label" : "Start CubeMX",
    "type" : "process",
    "command" : "csolution",
    "args": [
        "${command:cmsis-csolution.getSolutionPath}",
        "run",
        "--generator", "CubeMX",
        "--context", "BlinkTest.Debug+STM32H753ZITx"
    ],
    "problemMatcher" : []
}
````

The desired structure of the instruction is created by the `task`. However, `CubeMX `starts and after a moment CubeMX is terminated again. I have used `process `and `shell ` as `type`. In both cases the result is the same (termination of CubeMX). 

If you copy the generated instruction into the terminal and execute it in the terminal directly, CubeMX is not closed. I would be very grateful if I could get a tip on how to prevent CubeMX from quitting immediately.

My **second question** concerns the entry after `context`. Is there a possibility that I can get the current one for `context `using a `csolution-command`?

jkrech, 2024-06-13:
@FerdinandEn, thanks for this piece of feedback.
We plan to let people launch the generator from the CMSIS Outline View. The context will be selected based on the currently configured/active context set in line with the displayed outline view.

I did not find a command to find out about the active context. 

Can you confirm that CubeMX behaves differently when you launch the command via the terminal inside VSCode or outside or does CubeMX terminate there as well?

FerdinandEn, 2024-06-13:
@jkrech When I start the task with `Terminal-> Run Task...-> Start CubeMX` or `>Tasks:Run Task. -> Start CubeMX`, the problem occurs.

If I enter the instruction in the VSCode terminal, everything is fine. I use the instruction that is displayed by the `Run Task call` in the terminal. Copy and paste within the same terminal window. Then everything works.

jkrech, 2024-10-15:
@FerdinandEn we have released extension version 1.40 last week. 
You can launch a generator from the CMSIS outline view.
Under "Components" look for "Device:CubeMX" 

[image]

Click on the icon to launch STM32CubeMX.

ReinhardKeil, 2024-10-15:
This are the new packs with CubeMX suppport https://github.com/Open-CMSIS-Pack#stm32-packs-with-generator-support
More are coming soon

ReinhardKeil, 2024-10-18:
See also https://www.linkedin.com/feed/update/urn:li:activity:7252948507068665856/

jkrech, 2024-10-28:
I was not able to find a way to launch a process that does not terminate its child processes when the invoked tool (i.e. cbridge) returns/exits.

closed 2024-10-28 by jkrech

FerdinandEn, 2024-10-28:
With the new version. I can directly start CubeMX. It works fine.